bdring/Grbl_Esp32 #895: Using I2SO with SPINDLE_DIR_PIN and SPINDLE_ENABLE_PIN gives conversion error

Protoneer, 2021-05-12:
**What version of the firmware are you using?** Latest
**Is the problem repeatable?** Yes

**Under what conditions does the bug occur?**

Setting the machine file as follow:
```C++
#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_2
#define SPINDLE_ENABLE_PIN      I2SO(28)
#define SPINDLE_DIR_PIN         I2SO(29)
```

Arduino error
```
C:\Users\IT\AppData\Local\Temp\arduino_build_548812\sketch\src\Spindles\10vSpindle.cpp: In member function 'virtual void Spindles::_10v::deinit()':
C:\Users\IT\AppData\Local\Temp\arduino_build_548812\sketch\src\Spindles\10vSpindle.cpp:177:42: error: invalid conversion from 'int' to 'gpio_num_t' [-fpermissive]
         gpio_reset_pin(SPINDLE_ENABLE_PIN);
                                          ^
In file included from C:\Users\IT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5/tools/sdk/include/esp32/esp_sleep.h:19:0,
                 from C:\Users\IT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5/tools/sdk/include/esp32/esp_system.h:21,
                 from C:\Users\IT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5/tools/sdk/include/freertos/freertos/portable.h:128,
                 from C:\Users\IT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5/tools/sdk/include/freertos/freertos/FreeRTOS.h:105,
                 from C:\Users\IT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\cores\esp32/Arduino.h:32,
                 from c:\users\it\appdata\local\temp\arduino_build_548812\sketch\src\grbl.h:28,
                 from C:\Users\IT\AppData\Local\Temp\arduino_build_548812\sketch\src\Spindles\Spindle.h:44,
                 from C:\Users\IT\AppData\Local\Temp\arduino_build_548812\sketch\src\Spindles\PWMSpindle.h:24,
                 from C:\Users\IT\AppData\Local\Temp\arduino_build_548812\sketch\src\Spindles\10vSpindle.h:29,
                 from C:\Users\IT\AppData\Local\Temp\arduino_build_548812\sketch\src\Spindles\10vSpindle.cpp:27:
C:\Users\IT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5/tools/sdk/include/driver/driver/gpio.h:261:11: note:   initializing argument 1 of 'esp_err_t gpio_reset_pin(gpio_num_t)'
 esp_err_t gpio_reset_pin(gpio_num_t gpio_num);

```

The above error repeats for the SPINDLE_ENABLE_PIN and on the following functions virtual void Spindles::PWM::deinit()

Protoneer, 2021-05-12:
Currently it looks like these pins can only be set to an ESP pin and not to I2SO pins.

[https://github.com/bdring/Grbl_Esp32/blob/f3ca8b24467f7f39342ad4ff49f982b7f415cb29/Grbl_Esp32/src/Spindles/10vSpindle.cpp#L177]

Just looking for examples where I2SO were used. Might be able to make it usable...

MitchBradley, 2021-05-13:
I suspect that the calls to gpio_reset_pin are unnecessary.  The pinMode() call that follows each of them is probably sufficient.

Protoneer, 2021-05-13:
> I suspect that the calls to gpio_reset_pin are unnecessary in the ENABLE_PIN and DIR_PIN cases. The pinMode(,,,, INPUT) call that follows each of them is probably sufficient.
> 
> The case where gpio_reset_pin() might be necessary/helpful is for OUTPUT_PIN, because that is used with the LEDC controller to do PWM. In deinit(), gpio_reset_pin() would detach it from the LEDC controller.
> 
> So I would try removing the calls to gpio_reset_pin() for everything except *_OUTPUT_PIN . I2S0 cannot be used for OUTPUT_PIN anyway, because only GPIOs can be attached to the LEDC controller.

Thanks... We are making progress.

I was able to start with just:

```
#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_2
```

I removed the gpio_reset_pin() as suggested... and was able to compile with:
```
[MSG:PWM spindle Output:GPIO(2), Enbl:I2SO(28), Dir:I2SO(29), Freq:5000Hz, Res:1 3bits]

```

However the spindle_enable_pin does weird stuff... when GRBL is at idle it reads 0V when I run X1000 it ocilates between 2 and 3V.

For my coolant pin I used :
```
#define COOLANT_MIST_PIN        I2SO(30)
```

And that is super stable while in Idle or Run mode.

bdring, 2021-05-13:
The I2S0 pins were originally intended to driver stepper motors. They can also be used for on/off signals (no PWM).

We did not consider them for PWM spindles, but they should work Enable and Dir uses. The `gpio_reset_pin(...)` causes problems for that. It either needs to be removed for those pins or abstracted to be compatible with all pin types.

I commented out that function and ran this setup.

```
#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_26
#define SPINDLE_ENABLE_PIN      I2SO(24)
#define SPINDLE_DIR_PIN         I2SO(25)
```

It worked fine. Here is an M3S100 (10%) captured on my logic analyzer. The enable signal is stable on my setup.

[image]

MitchBradley, 2021-05-13:
The SPINDLE_ENABLE_PIN "oscillation" on I2SO(28) suggests a bad connection somewhere between the shift register chip and the measurement point.  Is there a load on the pin?  If the bad connection guess is correct, then the addition of a load will get rid of the oscillation and the voltage will always be 0.

Protoneer, 2021-05-13:
> The I2S0 pins were originally intended to driver stepper motors. They can also be used for on/off signals (no PWM).
@bdring  I agree and I will only be using the I2SO pins for on/off outputs.

@MitchBradley I will check the connection again. Would a 10K pull down be oaky for load?

MitchBradley, 2021-05-13:
10k pull down would be enough load to detect an open circuit.  You could also probe directly on the shift register pin.  Sometimes probing the pin will "fix" the problem if it is a poor solder joint between the pin and the board.  The probe pushes the pin down so it touches the pad better.  A fix like that will not last, though.  You need to reflow the solder to restore a connection properly.

Protoneer, 2021-05-18:
> The SPINDLE_ENABLE_PIN "oscillation" on I2SO(28) suggests a bad connection somewhere between the shift register chip and the measurement point. Is there a load on the pin? If the bad connection [...]

Spot on!!! My screw connector was a bit loose.

All is working!!! Thanks

closed 2021-05-18 by Protoneer