bdring/Grbl_Esp32 #896: Using I2SO pins for user_digital_pin_X does not

Protoneer, 2021-05-12:
**What version of the firmware are you using?** Latest

**Is the problem repeatable?** Yes

**Under what conditions does the bug occur?**

Using the following settings, the user_digital_pin_X      's does not get assigned.
Yet, COOLANT_MIST_PIN works as expected and is set as configured when restarting the ESP32

```C++
// I2S (steppers & other output-only pins)
#define USE_I2S_OUT
#define USE_I2S_STEPS
//#define DEFAULT_STEPPER ST_I2S_STATIC

#define USE_STEPSTICK   // makes sure MS1,2,3 !reset and !sleep are set

#define I2S_OUT_BCK             GPIO_NUM_22
#define I2S_OUT_WS              GPIO_NUM_17
#define I2S_OUT_DATA            GPIO_NUM_21

#define user_digital_pin_0      I2SO(24) //  m62 m63
#define user_digital_pin_1      I2SO(25) //  m62 m63
#define user_digital_pin_2      I2SO(26) //  m62 m63
#define user_digital_pin_3      I2SO(27) //  m62 m63

#define COOLANT_MIST_PIN        I2SO(30)
```

bdring, 2021-05-12:
`#define USER_DIGITAL_PIN_0` is case sensitive. This is all C language.

Protoneer, 2021-05-13:
> USER_DIGITAL_PIN_0

Thanks!!!  That sorted it.... :)

closed 2021-05-13 by Protoneer