bdring/Grbl_Esp32 #900: Ganged axes squaring with dedicated endstop inputs

lkarlslund, 2021-05-17:
I am currently migrating my Arduino based GRBL machine to an ESP32 GRBL machine, and have noticed some differences when using auto-squaring setups. My machine has a dual Y setup, with dedicated stepper drivers for each side of the machine and also dedicated end stops for each side.

Old:
- End stop inputs are "shared" across axes (in my case Y2 is put on Z)
- You can home multiple axes a once (as long as you do not collide with the shared axis), my sequence is Z then XY
- No unneeded stress on system, as everything is squared together when the machine is square to start with

New:
- End stop inputs are ambiguous across the squaring axis (Y and Y2 share input)
- You can't home a squaring axis at the same time as any other axis (so new sequence would need to be Z, then squaring Y and then finally homing X). This is way slower - yes I could home ZX as the first thing, but I'd prefer to ensure that the Z-axis is fully retracted before moving around in the XY plane.
- Some mechanical stress is put on the system, as each side of the squaring axis is homed individually after triggering end stop once

While the general idea about offering squaring on any axis, providing you have stepper driver outputs enough but with no extra input requirements, I think this is both well though out and scales nicely.

Unfortunately it has some drawbacks if you do have extra inputs available or wish to use the shared input mode of the older GRBL firmware:
- extra time is spent homing 
- mechanical stress put on the system every time you home the machine

I'd love the option to either share an input with another axis (old GRBL style) or use a dedicated input for this type of setup.

*Finally huge thank you goes out to everyone involved in creating this advanced fork of GRBL, developing PCBs and everything else that enables improved home grown CNC machines.*

jlirochon, 2021-05-17:
I am using [grbl-Mega-5X](https://github.com/fra589/). My machine has dual X and dual Y setup. 2 dedicated endstops by axis, nothing shared. I like the homing sequence.

I didn't try the shared endstop feature on Grbl_Esp32, but it seems a little weird to me a priori. Like @lkarlslund I have concerns about mechanical stress. I will experiment anyway, maybe the stress will be insignificant like the doc suggests.

In hindsight, the feature is clever and interesting, but if there are enough inputs, functionally, I think dedicated endstops should be supported as well (or should be the norm, shared endstops being an clever optimisation with eventual drawbacks).

bdring, 2021-05-17:
You can have 2 separate inputs per switch.

```C++
#define X_LIMIT_PIN ...
#define X2_LIMIT_PIN ...
```

This is a highly requested and popular feature for DIY machines that are not so rigid.

This feature assumes your axis has some slop and is prone to being out of square. In that case the sequence should not excessively stress the axis. You can set the level of backoff if stress is a concern. If your machine is rigid, you don't need to use the feature.

dpenney, 2021-05-25:
For an MPCNC with two steppers per axis would this be expected to work? At first glance it appeared that it was worrying for X but not Y, it would find the first Y switch and just use that, never driving the Y steppers individually.

#define X_LIMIT_PIN ...
#define X2_LIMIT_PIN ...

#define Y_LIMIT_PIN ...
#define Y2_LIMIT_PIN ...
```

closed 2021-05-26 by bdring