bdring/Grbl_Esp32 #903: Both side limit switches on each axis

karoria, 2021-05-19:
Hi @bdring @MitchBradley,

I want to implement both side limit switches on each axis with separate input to controllers so that machine doesn't get confused in case of any limit switch is pressed when we issue home command.
can I use direct pin or i2s? If possible, how about code changes?

Thanks in anticipation.

Regards,
Ravi

bdring, 2021-05-19:
You can only use ESP32 pins, because I2S is output only.

We are working on a new version of Grbl_ESP32 that will support this. It is likely to take a few more months. We do not plan to add it to the current version.

karoria, 2021-05-19:
Thanks for prompt reply.. I am glad to know it is in your roadmap. Thanks again.

closed 2021-05-19 by karoria