bdring/Grbl_Esp32 #905: Print bug report

solawc, 2021-05-21:
Version :The latest main branch

Use configuration:

- 1、enable hard limit

- 2、enable soft limit

- 3、use SDcard to print .nc file

report info:

[ESP220]/12qwe.nc

<Run|MPos:6.800,2.438,0.000|FS:1000,0|Pn:P|SD:0.17,/12qwe.nc>

<Run|MPos:17.263,15.337,0.000|FS:1000,0|Pn:P|SD:0.17,/12qwe.nc>

<Run|MPos:26.450,26.700,0.000|FS:0,601|Pn:P|Ov:100,100,100|A:S|SD:0.88,/12qwe.nc>

<Run|MPos:23.812,26.500,0.000|FS:0,0|Pn:P|Ov:100,100,100|SD:4.13,/12qwe.nc>

<Run|MPos:30.212,25.150,0.000|FS:1000,601|Pn:P|Ov:100,100,100|A:S|SD:9.68,/12qwe.nc>

<Run|MPos:30.225,23.987,0.000|FS:1000,601|Pn:P|WCO:0.000,0.000,0.000|SD:15.45,/12qwe.nc>

<Run|MPos:27.913,23.188,0.000|FS:1000,0|Pn:P|Ov:100,100,100|SD:17.95,/12qwe.nc>

<Run|MPos:27.650,23.188,0.000|FS:537,601|Pn:P|Ov:100,100,100|A:S|SD:21.08,/12qwe.nc>

<Run|MPos:23.938,23.263,0.000|FS:442,601|Pn:P|Ov:100,100,100|A:S|SD:27.21,/12qwe.nc>

<Run|MPos:34.388,21.225,0.000|FS:331,601|Pn:P|Ov:100,100,100|A:S|SD:28.92,/12qwe.nc>

<Run|MPos:31.725,21.112,0.000|FS:960,601|Pn:P|Ov:100,100,100|A:S|SD:31.18,/12qwe.nc>

<Run|MPos:25.913,20.825,0.000|FS:1000,0|Pn:P|Ov:100,100,100|SD:34.09,/12qwe.nc>

<Run|MPos:32.188,20.750,0.000|FS:1000,0|Pn:P|Ov:100,100,100|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|WCO:0.000,0.000,0.000|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|Ov:180,100,100|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|Ov:190,100,100|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|Ov:200,100,100|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|WCO:0.000,0.000,0.000|SD:35.64,/12qwe.nc>

<Idle|MPos:35.713,20.712,0.000|FS:0,0|Pn:P|SD:35.64,/12qwe.nc>

and ,

Grbl 1.1 ['$' for help]

[ESP220]/12qwe.nc

[ESP220]/12qwe.nc

<Run|MPos:29.575,6.575,0.000|FS:1000,0|Pn:P|Ov:100,100,100|SD:66.14,/12qwe.nc>

<Run|MPos:23.800,7.387,0.000|FS:1000,601|Pn:P|WCO:0.000,0.000,0.000|SD:70.20,/12qwe.nc>

<Run|MPos:29.850,6.062,0.000|FS:797,601|Pn:P|Ov:100,100,100|A:S|SD:74.19,/12qwe.nc>

<Idle|MPos:26.550,5.075,0.000|FS:0,0|Pn:P|Ov:100,100,100|SD:77.66,/12qwe.nc>

<Run|MPos:26.138,4.600,0.000|FS:1000,0|Pn:P|Ov:100,100,100|SD:82.34,/12qwe.nc>

<Run|MPos:27.150,3.500,0.000|FS:348,0|Pn:P|Ov:100,100,100|SD:83.87,/12qwe.nc>

<Run|MPos:27.075,3.525,0.000|FS:555,601|Pn:P|Ov:100,100,100|A:S|SD:88.57,/12qwe.nc>

<Run|MPos:22.700,3.225,0.000|FS:1000,0|Pn:P|Ov:100,100,100|SD:91.98,/12qwe.nc>

<Run|MPos:6.275,1.000,0.000|FS:407,0|Pn:P|SD:91.98,/12qwe.nc>

<Run|MPos:0.950,12.363,0.000|FS:1000,601|Pn:P|Ov:100,100,100|A:S|SD:95.74,/12qwe.nc>

<Run|MPos:1.025,28.862,0.000|FS:1000,601|Pn:P|SD:96.03,/12qwe.nc>

<Run|MPos:15.675,26.625,0.000|FS:1000,601|Pn:P|WCO:0.000,0.000,0.000|SD:96.82

I think it's strange that in the printing process, it should be run, but sometimes it will appear idle, which is very difficult to understand

solawc, 2021-05-21:
After printing stopped, the laser head did not turn off, but remained on，But through the command to continue to control, such as moving, it is still normal action

bdring, 2021-05-27:
I do not understand the problem you are having? 

- Are you in laser mode
- Is there an M5 command at the end of you gcode?

https://github.com/bdring/Grbl_Esp32/wiki/Laser-Mode