bdring/Grbl_Esp32 #907: Homing error/ adding z axis without endstop and gui not working

nukes1810, 2021-05-24:
Hello
I am usng espduino32 with the cnc shield 3.0 to use on a diode laser engraver. i have compiled the firmware and everything seems to work fine except the homing on light burn i get alarm 8 which im guessing is caused because it is still trying to home z-axis first. I used the pen_laser machine and on the defaults, i set it as follows

// Grbl generic default settings. Should work across different machines.
#ifndef DEFAULT_HOMING_CYCLE_0
#    define DEFAULT_HOMING_CYCLE_0 (bit(X_AXIS) | bit(Y_AXIS))
#endif

#ifndef DEFAULT_HOMING_CYCLE_1
#    define DEFAULT_HOMING_CYCLE_1 0
#endif

// ========= AXIS MAX TRAVEL ============
#ifndef DEFAULT_HOMING_CYCLE_0
#    define DEFAULT_HOMING_CYCLE_0 (bit(X_AXIS) | bit(Y_AXIS))
#endif

#ifndef DEFAULT_HOMING_CYCLE_1
#    define DEFAULT_HOMING_CYCLE_1 0

And i wand to have the Z axis to use on my rotory without an endstop so can i just add the z axis like so on the pen_ laser 

//#define PEN_LASER_V1
#define PEN_LASER_V2

#define X_STEP_PIN              GPIO_NUM_26
#define X_DIRECTION_PIN         GPIO_NUM_16

#define Y_STEP_PIN              GPIO_NUM_25
#define Y_DIRECTION_PIN         GPIO_NUM_27

#define Z_STEP_PIN              GPIO_NUM_17
#define Z_DIRECTION_PIN         GPIO_NUM_14

#define STEPPERS_DISABLE_PIN    GPIO_NUM_12

and finally, when i go to the WebGUI it's telling me that the index page is mussing how do i add that. so sorry for the lengthy questions but I have no background on this and somehow i pieced everything together but i can't seem to solve these issues.

Thank You

MitchBradley, 2021-05-24:
Here is the documentation for the WebUI index page.  https://github.com/bdring/Grbl_Esp32/wiki/ESP3D-Web-UI-for-Grbl_ESP32#first-connection .  You can learn a lot by reading the wiki.

nukes1810, 2021-05-24:
Thank you for that , apologies I didn't read the wifi setup that much but I did went over the homing and endstop and can't find an answer on why I can't get rid of z homing

MitchBradley, 2021-05-24:
Alarm 8 means that the limit switch did not turn off during the pulloff move.  Maybe your pulloff distance is too short.

bdring, 2021-05-26:
See this section of the Wiki. It could be the switch logic is inverted.

https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-Limit-Homing-Switch#first-tests

nukes1810, 2021-05-27:
thank you for all the help i did namaged to get x to home properly but Y does not home it moves few mm towards home but does not home at all and software is saying its at home position

bdring, 2021-05-27:
Can you post the status response with the Y switch in the untouched and touched state.

It is possible your switch is reporting as touched when not touched.