bdring/Grbl_Esp32 #908: Serial communication to another Arduino for PID spindle control

CNCNoob, 2021-05-25:
Hi guys, first off I want to say thank you to the key contributors of this project, you guys have done an amazing job so far, and I know you will continue to do so! the GRBL_ESP32 setup has made many features possible compared to the less functional Arduino version.

Lately, I've been tinkering with the idea of having a PID closed-loop controller for the spindle of my CNC, as it would be nice to have a beautiful and consistent surface which is only possible when having a constant (and accurate) spindle speed, I've developed some Arduino code, which works reasonably well (altho it needs fine-tuning) but this obviously needs to link up to the esp32 in some way.

My thoughts were to write up some custom spindle .h and .cpp files, which would send the required data over a simple uart protocol. Ideally, this would only need 1 pin, but I understand that 2 will probably be needed (one for tx pin, and the other for CLK) Is there some way I can make use of the already implemented VFD spindle code for the communication? as I noticed that the "HuanyangSpindle.cpp" used some simple communication techniques, but I don't know if I'm going to mess anything in the main code up, + where do I define the pinouts? (or are they're only fixed uart outputs?) below are the simple communication techniques that I'm wondering if I can implement:

```
data.tx_length = 6

data.msg[3] = (value >> 8) & 0xFF;
data.msg[4] = (value & 0xFF);
```

This feature could be a massive benefit to the Esp32 as it would reduce the need for PID closed-loop to be built into the esp32-grbl code itself, but instead would be a VFD-like controller which would be customizable for any spindle.

MitchBradley, 2021-05-25:
Emulating the Huanyang protocol would be a good approach.  It is built on top of the Modbus protocol.  The physical interconnect is a UART with an RS-485 adapter to convert it to a half-duplex twisted pair.  Grbl_ESP32 currently uses UART2 for that purpose.  It might become configurable at some point, but for now there is little reason to change it.  If you emulate Huanyang, the existing software should just work.  
In principle, you could use a direct UART instead of going to the trouble of conversion to and from RS485, but in the case of a spindle, there is quite a lot of benefit from the noise immunity of RS485's differential signaling over twisted pair.  Spindles can be very noisy at the electrical level.

CNCNoob, 2021-05-25:
Thank you MitchBradley for the fast reply!
I did some research on the protocol for the Modbus protocol, but it seemed very overcomplicated for the spindle controller which I am making, which just needs a spindle RPM value (and possibly a direction).

But I figured out that the 3rd hardware uart port was not used in the GRBL code, and therefore have used Serial2.print to send the RPM values to the Arduino. I plan to keep working on this to make it more sophisticated. Maybe I will add a while loop that waits for the Arduino spindle to send back the "ok" signal, saying that it has reached its target RPM. I have implemented this code into a custom spindle file.

I'm just wondering though, as I have not used GitHub very much when I have completed my code, do I start a pull request, so that other people can test/check my code?

Thanks

MitchBradley, 2021-05-25:
There are pull request guidelines on the wiki.  Your chances of getting your PR accepted will be much greater if you do as I suggested - but you might not need to PR anything because if you emulate one of the existing VFDs the existing code will just work.  If you go your own route you are very likely to collide with other work that we have planned.  Serial2 will not work.  We do not use HardwareSerial because of bugs that were introduced in a recent version of the Arduino framework.  We also have some ongoing work that involves UART 2.  The modbus protocol is not that difficult.  You just have to accept a few structured messages and construct suitably-formatted replies.  You might be able to implement do-nothing versions of the requests that your device does not support.  I guarantee you that emulating a modbus VFD will be easier than getting us to accept a PR that collides in any way with what we are doing - which is pretty much inevitable given that you are new to the code base and not yet plugged in to the ongoing development.

MitchBradley, 2021-05-25:
If you look at existing VFD spindle drivers like Spindles/H2ASpindle.cpp or Spindles/YL620.cpp,  you will see that the commands we use are init, which gets the maximum RPM, get_current_rpm(), direction_command which sets the direction or disables, set_speed_command(), get_current_direction() and get_current_rpm().  That sounds like exactly what you are talking about.  We already have all the code to wait for spin up, spin down and whatnot.

CNCNoob, 2021-05-25:
Thanks again MitchBradley, that makes a lot of sense! I may as well just go along with it then! I'll look into the Modbus a bit further, so as to keep away from bugs! I'm sure you guys know what you're talking about, so it'd be best for me to follow along!

closed 2021-05-25 by CNCNoob