bdring/Grbl_Esp32 #910: Problem with all 6 axes

Juank23, 2021-05-29:
With any configuration it continues to show me the 6 axes, previously one configured the number of axes and it was fine thanks

bdring, 2021-05-29:
You probably have a problem in your machine definition. Can you double check that you have the right one in Machine.h.

Also post the machine file and boot messages here.

https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages

Juank23, 2021-05-29:
buen dia, gracias por responder, aparentemente todo esta bien en el archivo cualquier maquina que defino asi tenga N_AXIS 3
sigue mostrando los 6 ejes, aparte de eso hay un problema que note por wifi,  le envio el codigo que dura 10 minutos mecanizando
y despues de unos 4 minutos se reinicia gracias

MitchBradley, 2021-05-29:
What do you mean by "it continues to show me the 6 axes".  What is "it" ?  Are you using a GCode sender to interact with Grbl_Esp32?  If so, which one?  What is the specific indication of 6 axes?  Please include a screenshot.

Juank23, 2021-05-29:
[image]
[image]
[image]

bdring, 2021-05-29:
What is the response if you send the ? character.

like...

<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>

MitchBradley, 2021-05-29:
MAX_N_AXIS is the maximum number of axes (6) that the software can possibly support, not the number of axes that are currently in use by a particular configuration.  The number of axes that are actually in use can range from 3 to 6

Juank23, 2021-05-29:
Thank you for your prompt responses ... so one defines 3 or 4 axes continues to show 6 but only the axes that one defines are used ???

MitchBradley, 2021-05-29:
The GRBL settings list - items like $130 .. $135 - is based on MAX_N_AXIS, not on the currently configured number of axes.  We are working on a new configuration mechanism that is based on a hierarchical text description of the machine.

In a status report response, only the configured axes will be reported.  Attempts to use an axis letter beyond the configured number will fail - so if you have 4 axes configured and you try to use the B axis, that will not work.

Juank23, 2021-05-29:
bdring, well I am really testing it by IP address and it connects very well and works but in a few minutes of being machining it disconnects apparently the controller is restarted, things that does not happen through bluetooth or connected directly by usb

Juank23, 2021-05-29:
MitchBradley, you are not understanding me, previously with this code one would select the number of axes that one was going to use in only 
[image]
showing the selected axes .. now notice that if one selects 3 axes it continues to show the 6 axes

MitchBradley, 2021-05-29:
The code changed, and will change again when we finish the new settings scheme.

Juank23, 2021-05-29:
ok thanks, but that does not affect the operation of the controller ??

MitchBradley, 2021-05-29:
Do you mean the sender?  In most cases, senders are not responsible for machine setup.  The protocol for sending GCode will not change, and the few settings that senders typically use directly - like $10 and sometimes $13 - will continue to be supported.  The old $nnn format is too restrictive to support all the new features that we have developed. If you want a preview of what is coming, look at this page https://github.com/bdring/Grbl_Esp32/wiki/YAML-Config-File .

With the new scheme, recompiling the firmware will not be necessary.  You can configure everything, including the pin usage, simply by loading a text file into the SPIFFS filesystem.

Juank23, 2021-05-29:
Well, I have tried most of the codes in an esp32 that I have, and it works well for bluetooth .. For wiffi and ip address it works for a few minutes and the controller restarts, as if receiving so much code it would block and restart, that's why I thought which could be due to the number of axes that appear so defined 3

Juank23, 2021-05-29:
[image]

MitchBradley, 2021-05-29:
You violated the primary rule of bug reporting - instead of telling us exactly what you did, what you expected to happen, and what actually happened, you guessed at what might be going wrong and reported your guess.

MitchBradley, 2021-05-29:
Connect a terminal emulator to the USB serial port and show us the messages that appear on the serial port when it restarts.

Juank23, 2021-05-29:
MitchBradley Sorry, I realized that error so far with the tests I'm doing, thanks for your answers

Juank23, 2021-05-29:
[image]

MitchBradley, 2021-05-29:
This looks similar to an out of memory or task stack overflow.  Can you use the Arduino exception decoder?

Juank23, 2021-05-29:
Sorry, and how is that done ???

MitchBradley, 2021-05-29:
[image]

MitchBradley, 2021-05-29:
Search is your friend

Juank23, 2021-05-29:
[image]

MitchBradley, 2021-05-29:
[image]

That is a stack trace

Juank23, 2021-05-29:
[image]

MitchBradley, 2021-05-29:
Are you using Telnet?

Juank23, 2021-05-29:
I am using the configuration that the code brings, I have not configured anything
ip 192.168.0.1
[image]

MitchBradley, 2021-05-29:
What machine file are you using?  Showing the source code for WifiConfig.h is pointless because that never changes.  The configuration is in Machines/<your_machine>.h and in the settings shown when you type $S. 

This part of the backtrace suggests that there is a telnet program running on some other machine that is connecting to Grbl_Esp32.  
[image]

Juank23, 2021-05-30:
Good morning, friend, just download the firmware and in machine.h #include "Machines / 3axis_v4.h", or do you have to do some other configuration? Thanks