bdring/Grbl_Esp32 #913: Request for advice on GUI part for A, B, C axis

cimygy, 2021-06-03:
** Please describe the function to be implemented **

Control method when adding A, B, C axes

**Why do you think Grbl_ESP32 will be an improvement? **

A, B, and C axes can be added, but it is difficult to control

**Why do you need this feature? **

I can add A, B, C axes, but I can't get the JOG and origin

**Is this feature visible to many users? **

I am trying to make a 4 axis CNC of XYYZA type.
I'm trying to use WEB UI or Candle.
I don't know how to manipulate the A axis at this time.
I want to do JOG motion, origin capture, and current position indication.
Please advise on how the GUI considers the A axis.

bdring, 2021-06-03:
You control the ABC axes in the WebUI by changing the axis in the drop down list. 

[image]

Senders like Candle would have to figure out how they want to do it on their own.

cimygy, 2021-06-04:
Ah! It has already been implemented in the Web GUI.
Sorry for not knowing the details and asking the question.
Please understand that I am still learning to implement.

I'm still studying, so I don't know the depth, but it's a very interesting project.
Thank you for making such a great project.

closed 2021-06-04 by cimygy