bdring/Grbl_Esp32 #916: MPCNC Dual Endstops

AlexHern1971, 2021-06-10:
Question about wiring dual endstops for an MPCNC, just received my board and I only ordered one 4x input module. Can I do auto squaring if the two endstops on an axes are wired in parallel? Should I order another input module so I can wire each endstops individually?

bdring, 2021-06-11:
Currently the code internally combines the switches if you have multiple inputs per axis. If you have simple mechanical switches there is no advantage to multiple inputs.

There is more information here about how to wire them.

https://github.com/bdring/Grbl_Esp32/wiki/Motor-Ganging-and-Axis-Squaring#switches

In the future (months, not weeks) we are planning to add features to take advantage multiple switches.

AlexHern1971, 2021-06-12:
Thank you for the clarification.

closed 2021-06-12 by bdring