bdring/Grbl_Esp32 #917: GCode comments are only sent to the serial interface even when connected using WiFi

HuubBuis, 2021-06-14:
What version of the firmware are you using?
[VER:1.3a.20210424:]
[OPT:PHBW]

Is the problem repeatable?
Yes

Under what conditions does the bug occur?
Sending Gcodes having comments like "G91 X10 F100 (MSG,MyComment)"

I have checked the code and found that Gcode Comments are sent to the CLIENT_SERIAL interface.
According to the [Wiki:](https://github.com/bdring/Grbl_Esp32/wiki/Supported-GCodes#comments), it should be send to all open interfaces. 
I changed it to CLIENT_ALL and now it is working OK.

```C++
void report_gcode_comment(char* comment) {
    char          msg[80];
    const uint8_t offset = 4;  // ignore "MSG_" part of comment
    uint8_t       index  = offset;
    if (strstr(comment, "MSG")) {
        while (index < strlen(comment)) {
            msg[index - offset] = comment[index];
            index++;
        }
        msg[index - offset] = 0;  // null terminate
        grbl_msg_sendf(CLIENT_ALL, MsgLevel::Info, "GCode Comment...%s", msg);
    }
}
```

I noticed that error messages when using button macro's, are also only send to the CLIENT_SERIAL interface.  I am not sure if this is intended.
I changed it to CLIENT_ALL in my [fork](https://github.com/HuubBuis/Grbl_Esp32) because I normally connect using BT or WiFi.

```C++
system.cpp
void __attribute__((weak)) user_defined_macro(uint8_t index) {
    // must be in Idle
    if (sys.state != State::Idle) {
        grbl_msg_sendf(CLIENT_ALL, MsgLevel::Info, "Macro button only permitted in idle");
        return;
    }

    String user_macro;
    char   line[255];
    switch (index) {
        case 0:
            user_macro = user_macro0->get();
            break;
        case 1:
            user_macro = user_macro1->get();
            break;
        case 2:
            user_macro = user_macro2->get();
            break;
        case 3:
            user_macro = user_macro3->get();
            break;
        default:
            return;
    }

    if (user_macro == "") {
        grbl_msg_sendf(CLIENT_ALL, MsgLevel::Info, "Macro User/Macro%d empty", index);
        return;
    }

    user_macro.replace('&', '\n');
    user_macro.toCharArray(line, 255, 0);
    strcat(line, "\r");
    WebUI::inputBuffer.push(line);
}
```

bdring, 2021-06-16:
Sending to CLIENT_ALL is acceptable. Can you submit a PR against the devt branch?

https://github.com/bdring/Grbl_Esp32/wiki/Pull-Request-Guidelines

HuubBuis, 2021-06-16:
I will change and test using the devt branch and submit a PR