bdring/Grbl_Esp32 #919: I can't connect to lightburn using this firmware

sorrowfeng, 2021-06-18:
**Please describe the feature you would like implemented**

**Why do you think this would improve Grbl_ESP32?**

**What do you need the feature for?**

**Will this feature appear to a lot of users?**

richardwpaul, 2021-06-18:
I have no problem communicating with LIGHTBURN.

Rick Paul
mrp email

MitchBradley, 2021-06-18:
There is not nearly enough information to help us debug the problem.  We need information like which platform you are using (Mac, Windows or Linux), which serial port you have tried to connect with, how you know that is the correct port, whether you have tried to connect to that port with other senders, screenshots of any error messages, and perhaps other information.

sorrowfeng, 2021-06-19:
I am using this branch, https://github.com/bdring/Grbl_Esp32/tree/Devt

The model selected in lightburn is GRBL, and when I tried to connect to lightburn, it immediately disconnected, and here is the console output.

`Waiting for connection...
�
[MSG:ESP32 SDK: v3.3.5-1-g85c43024c]
[MSG:Using cpu_map...CPU_MAP_SPI_DELTA]
[MSG:TMCStepper Init using Library Ver 0x000701]
[MSG:Machine Trinamic Setup]
[MSG:X Trinamic driver test passed]
[MSG:Y Trinamic driver test passed]
[MSG:Z Trinamic driver test passed]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Local access point GRBL_ESP started, 192.168.4.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
Grbl 1.1f ['$' for help]
G0
G0
G0
G0
G0
G0`

sorrowfeng, 2021-06-19:
> There is not nearly enough information to help us debug the problem. We need information like which platform you are using (Mac, Windows or Linux), which serial port you have tried to connect with, [...]

I'm using Win10, the port is fine, I can interact normally with other serial software, the model selected in lightburn is GRBL, when I try to connect to lightburn, it immediately disconnects, here is the console output.

sorrowfeng, 2021-06-19:
I use the code from the main branch to write to my arduino esp32 board, and after connecting, it also disconnects.
`Waiting for connection...
Waiting for connection...
[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:Using machine:Test Drive - Demo Only No I/O!]
[MSG:Axis count 3]
[MSG:Timed Steps]
[MSG:Init Motors]
[MSG:No spindle]
Grbl 1.3a ['$' for help]
G0
G0
G0
G0`

sorrowfeng, 2021-06-19:
> I have no problem communicating with LIGHTBURN. Rick Paul mrp email
> […](#)

I used the main branch of the code to write to my arduino esp32 board, which disconnected immediately after connection, and here is the console output.
`Waiting for connection...
Waiting for connection...
[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:Using machine:Test Drive - Demo Only No I/O!]
[MSG:Axis count 3]
[MSG:Timed Steps]
[MSG:Init Motors]
[MSG:No spindle]
Grbl 1.3a ['$' for help]
G0
G0
G0
G0
G0
G0`

NairoDorian, 2021-10-30:
News?

bdring, 2021-10-30:
I have Lightburn 9.10. and it connects for me. I posted a screen cast on Discord

[Discord Invite Link](https://discord.gg/7uKg7RXwha)

[Link to post](https://discord.com/channels/780079161460916227/903991615746617364/903992207588069378)

closed 2021-12-31 by sorrowfeng