bdring/Grbl_Esp32 #922: host SD as http or ftp server

karoria, 2021-06-19:
Hi @bdring @MitchBradley @atlaste @luc-github 
I am developing a proprietary software for our machines which run on grbl_esp32. Running a program from SD is already implemented with $SD/* commands. I want to incorporate hosting SD card files just like WEBUI, where I can upload and delete files also. I couldn't figure out from webui code how it actually works. Please guide me.

luc-github, 2021-06-19:
WebUI does not use `$SD/*` commands, just http queries

karoria, 2021-06-19:
I can understand that it should be possible via network only, not by direct serial communication. But I am not sure how to send http or ws requests. I am using socket.io for my project (rpi as server) and I want my esp32 to be connected to it as client and communicate to it. I need some example code or method to use. Thanks for time.

luc-github, 2021-06-19:
all sources code for grbl_esp32 and for webui are available, what better examples would you need ?

karoria, 2021-06-19:
I am not sure about which part of your code is required for my purpose. Secondly, I am not sure if I host esp32 server (for sd files) and also host my app server on rpi on same LAN, how to communicate between both servers.

MitchBradley, 2021-06-19:
WebUI uses HTTP POST to the URL  http://<grbl_esp32_ip>/upload

To understand the internals of the POST request you can look at the code in ESP3D-WEBUI/www/js/{files.js,http.js), or the code in Grbl_Esp32/src/WebUI/WebServer.cpp, or sniff the packets with something like WireShark.

karoria, 2021-06-19:
Thanks. This will help me. Closing now. If required, I will reopen.

closed 2021-06-19 by karoria