bdring/Grbl_Esp32 #926: Having trouble with YL620 Spindle Setting

Mawg1, 2021-06-24:
Is anyone else having an issue setting up the YL620 spindle?  I am using the latest push from April 25th.  I have tried defining it in the Machine file and setting it with the $Spindle/Type setting.  Both ways result in the setting not sticking and I get a readout of $Spindle/Type = ???  

[image]
 

Any suggestions?

MitchBradley, 2021-06-24:
There is a bug in the code.  The YL620 spindle is not included in the list of spindle type names.

Mawg1, 2021-06-24:
Thanks for the fast response.  Is the bug in the YL620 code, or in the spindle type code?  (Meaning is this something I can track down easily?)

Mawg1, 2021-06-24:
Sweet, thanks!