bdring/Grbl_Esp32 #927: I wanna rewrite the whole thing ...

phorton1, 2021-06-27:
Seriously.  I am _this_ close to starting it.

More later, but although I **love** ESP32_GRBL, and am making what I think may be substantial additions to it, I absolutely **hate** the codebase itself.

You don't know me but I come from almost 50 years of programming experience including some of the world's most popular (large, integrated, and/or web-based) code bases.

I wanna rewrite the whole thing as proper C++.   And get (tf) rid of Grbl.h as it presently stands.  The visibility of dozens, nay hundreds, of global methods and variables is absolutely horrible.  The naming conventions are thwarted.   The current code factoring and directory structure is entirely misleading.

The whole thing is not conducive to writing good additional code.

The longer such a rewrite is put off, the worse the problem will get.

I started off by simply wanting to implement a VPlotter (port the maslow kinematics, which I have done) and then I started adding a TFT user interface using LVGL and that's when I started down the slippery slope of trying to understand how it all really works.

My idea is to fit an intimate (not just a serial) client UI into the system.  After all, such a UI would run ON the ESP32 and I have access to all those global, right ???

:-)

None of my codebase is public at this point.  It is a copy (not a branch) of this codebase. But I am interested in discussing it.   Suggestions for the proper forum are welcome.

I am about 10 pages into writing a 400 page set of documents to myself describing how ESP32_GRBL currently works, lol.   

I *may* be publishing that, and a stab at an initial **architecture** for a well designed system in the future.   Or I may just **do** it.

I dunno.

But (to **bdring** primarily) I just wanted to reach out and say *howdy* and see if anyone else thinks the code is a mess, and valuable enough to rewrite properly.

Or might be open to talking about it.

I could not find bdring's email address.

So, lemme know.  I'm not sure how I'lll get replies.  I'm github/phorton1.  At least I should get one reply to this message by filing an "issue" on the codebase itself.

So, thanks bdring.  It's really a monumental effort you are doing and I have been on it 24/7 for about 3 months now.   

I look forward to any replies.

- Patrick

phorton1, 2021-06-27:
I went ahead and made my codebase public so that you can see
[my readme file](https://github.com/phorton1/Arduino-grbl_esp32#readme).

Please don't laugh, and more importantly, please don't get pissed off in a knee jerk reaction!!

I'm serious about this.  I think it could be on every CNC coming out of China next year.

Peace!

MitchBradley, 2021-06-27:
We agree with many of your sentiments and have been working diligently toward resolving architectural issues.  The YamlSettings branch has our latest code.  Look at its revision history to see all the stuff we have done.  It is still in testing, but will become the production version at some point.  Our initial focus is to eliminate the need for most users to recompile just to change basic machine configuration things.

We have already eliminated nearly all of the ifdefs that used to be present in the core code.  The only ones that remain are ones to completely disable Wifi and/or Bluetooth - which we want to do because the support libraries for those functions are enormous and we want the ability to reclaim that space for special applications - and some DEBUG ifdefs that guard very-rarely-used debugging statements.

Our approach has been to work from the outside in, focusing first on creating a class structure for pins and I/O devices, and making those configurable at runtime by reading a structured text file whose syntax is a subset of YAML.  As that stabilizes, or when the need arises, we touch core components.

One core component that we have started on is protocol_exec_rt_system().  The mass of nested ifs that process events has been replaced by a much simpler flow.  The next step for that module is to replace all the real time event flags with an event queue so protocol_exec_rt_system() dequeues an event and dispatches with a switch.  Every such change requires careful study of the interactions with other subsystems.  As you note, the interactions between modules are often poorly structured, but fixing them requires very careful analysis and testing, because the interactions are there for a reason and any improved structure must account for their purpose.

The client code is also a mess.  I have a patch to create a proper Client class, but I have held it back because the other work is more urgent.  The Client class patch will probably hit the YamlSettings branch in the next week or two.

We can use additional developers, but only if they can work as part of a team.  The tone of your rants is not encouraging on that front.

We are moving away from the Arduino IDE because it is so limiting.  Each of us main developers initially resisted the switch to platformio, but after trying it, cannot stand Arduino anymore.  It is not correct that platformio can only be used from VSCode. platformio is primarily a command line system that is written in Python.  It happens to be especially well integrated into VSCode, so VSCode users can install it with a few clicks, but the claim that VSCode is required is absolutely incorrect.  I do most of my editing in EMACS and can do platformio compiles inside an EMACS shell window or from any other shell.  Bart uses VSCode, while Stefan uses Visual Studio Community Edition.  There are good platformio integrations for many popular editors like Atom and vim - but you don't have to use an integration, as platformio works fine from the command line.

I too hate having to install gigabytes of stuff - but there is no getting around the need to install a toolchain and the necessary libraries.  platformio is extremely good at automating the installation of those things; better at it than any other system I have used in my 40+ years of professional programming.  Installation of the platformio core takes 60 seconds - I just reinstalled it from the instructions at https://docs.platformio.org/en/latest/core/installation.html#id30 and timed it.  It assumes that you already have python on your system - but if you are using esptool.py, you must have python, so that is not a new dependency.

One thing that is really helpful about platformio vs Arduino is the ability to inspect and debug inside Arduino library code.  With the Arduino IDE, the libraries are precompiled binaries so looking inside them requires extra work.  With platformio, the Arduino libraries are present as source code on your system, where you can easily look at the code and make changes or add printfs to trace what is happening.  I have used that facility several times to track down subtle bugs.  One particularly nasty bug was caused by an "improvement" in the Arduino serial port driver that resulted in bad GCode execution because the new version of the driver silently dropped characters.

phorton1, 2021-06-28:
Mitch,

Thank you for the prompt reply and it's nice to meet you  :-)

My name is Patrick.

I will have to look at the YamlSettings branch and try to get a sense of the wind direction.

I apologize for what might have seemed an unruly tone.   The fact of the matter is that I understand that GRBL itself was written for an arduino class processor, and that you and the other developers took it on and successfully completed a port of it to the ESP32, which is, in and by itself, a major accomplishment.  My comments are not meant in disparagement, but rather more as a rally cry for the future.

After all, I am using your code!   That in and by itself is the highest compliment I know how to pay another programmer :-)

I really *was* (maybe still am) at my wits end, though, with regards to trying to understand it's encapsulation, and hence, in my darker moments, I have seriously toyed with just rewriting it ... actually just re-factoring it in a big major way ... to some code I feel comfortable with.

I've been known to rewrite entire codebases over a weekend :-)

I will settle down :-)   And I hope at some point to be able to at least show you some interesting things, like Youtubes of my vPlotter thingee, and/or a demo of a TFT touch screen UI for GRBL_ESP32.  

Honestly, I wasnt expecting as meaningful and cognizant a response in such a short time frame ... so I have to also give you kudos for presence and attentiveness ... and attention to detail ... i noticed that, though, while reading many of your blog posts, that you answer questions, even when they are pedantic in nature .. so you're a kind person.  I like that.

I hope maybe with these additional words, that you'll take my post not so much as complaining about YOUR code, but rather as serious encouragment towards your goals.  And perhaps, just perhaps, there might be some way I could contribute to the mainline branch at some point.

And, btw, you are free to delete this issue if possible.  It was really only a way to start a conversation with you.

So, please don't take my rants too seriously !!

I will definitely have a look at the yaml branch, and take to heart your hard earned words regarding the Arduino IDE.   Some consideration is due.  I'm about to rebuild a new personal machine (yet again, 3rd in 3 months) that just came in the mail last week, so that might be a good time to move on, or sideways, from the Arduino IDE. 

There's a lot more, but I think that's plenty for now.

Have a nice Sunday night, and I'll TTYL!

Peace!
- Patrick

phorton1, 2021-06-28:
lol, you're an old guy too!

MitchBradley, 2021-06-28:
The problem with adding GUI stuff to Grbl_ESP32 is code space and pins.  We have been designing a way to hook up an auxiliary MCU, possible a super cheap STM32F040, using a couple of pins.  Not only could it host a graphical display, it could also serve as a cost effective input expander.

closed 2021-07-12 by bdring