bdring/Grbl_Esp32 #928: Problem command X and Y axis Polar Coaster V2

Project3334, 2021-06-30:
We have a problem to control the Polar coaster v2.

The X axis works, but if we give a command to the Y axis it doesn't work.
Also the Z axis from the servo motor isn't working.

If we give a command to the Y axis the X axis receives a command.

EsserPrototyping, 2021-07-09:
Hi, maybe this is the same problem I found.. (also a polar coaster, but I don´t think it´s related to polar only).

If you are using I2S and
#define I2S_OUT_NUM_BITS  16 and 
#define DEFAULT_STEPPER ST_I2S_STREAM

You´ll get wrong pulsing for (at least) DIR and the wrong motor moving. (On my oscilloscope it looks like the update of the shift registers is not performed right).

However when I #define DEFAULT_STEPPER ST_I2S_STATIC and 16 bits, it´s working fine.
If you´re not using 16 bits, it should work fine too, but I don´t tested it.

Project3334, 2021-07-11:
Hi @EsserPrototyping

in which file did you changed this? 
We found more files where the keyword i2s is written.
In a few documents the #define DEFAULT_STEPPER ST_I2S_STATIC is already implemented, but disabled by //.

To summarize and please correct me if I'm wrong:
We only have to define:
#define I2S_OUT_NUM_BITS 16
#define DEFAULT_STEPPER ST_I2S_STATIC

and disable:
#define DEFAULT_STEPPER ST_I2S_STREAM

EsserPrototyping, 2021-07-11:
First, it only applies if you´re using I2S. If not, i don´t think this will help.
If so.. are you using only two shift registers? If yes you should only need 16 bits. But 32 bit should run fine out of the box, if i´m right.
What i mean is, that the combination of 16bit and STREAM seems to be broken.
It worked for me by just putting
#define I2S_OUT_NUM_BITS 16
#define DEFAULT_STEPPER ST_I2S_STATIC
in my configuration.
To give a little hint on that.. when i checked it with my oscilloscope, i could see that the DIR output of the reigsters was not asserted at the right times (for example it only updated the DIR pin when i moved another axis).