bdring/Grbl_Esp32 #929: Finding GRBL ESP32 on the network

BarbourSmith, 2021-06-30:
Hello, 

I'm looking to build a quick and easy way for folks to find the IP address of of an ESP32 running GRBL on their networks. Something like findmyGRBLESP32.com which gives you a link to the ESP's IP address. That way folks won't need to worry about opening a serial connection ever which I think can be intimidating to non-programmers. 

My first thought was to use the mDNS system, but the more I look into it the more it seems platform, browser, and network preferences dependent which could cause confusion.

Now I'm thinking it might be possible to just run through and check all of the addresses in the 192.168 family of IP addresses. 

I doubt there is a way to do it which will work on 100% of network setups and devices, but I think it should be possible on most basic setups.

Has anyone done any work on this already? Does anyone have any suggestions for which request to make of each IP address to see if it is an ESP32 running GRBL?

bdring, 2021-06-30:
@luc-github 

Do you have any ideas about this?

MitchBradley, 2021-06-30:
Grbl_Esp32 already supports SSDP https://en.wikipedia.org/wiki/Simple_Service_Discovery_Protocol .

mDNS lets you find the IP address of one Grbl_Esp32 instance if you happen to know the hostname.  The default hostname in the mdns domain is grblesp.local but that is configurable.  If you wanted to search for hostnames, you would need to adopt a name schema so your search program could loop over a list of names.

IP address scanning seems like a bad idea to me, given the existing support for SSDP.  That wheel has already been invented.

MitchBradley, 2021-06-30:
You should only use SSDP within a local network.  Opening up SSDP from the outside is vulnerable to DDOS attacks.  Most routers block incoming SSDP automatically.  IP address scanning from outside would not work anyway, because of the way that unroutable subnets like 192.168.x.x work.

BarbourSmith, 2021-06-30:
SSDP does look nice but as far as I can tell it can't be called from within a browser. I could create executables for different operating systems which would use SSDP, but my guess is for most people it wouldn't be worth the trouble of installing a program.

I think to be a useful tool it would need to be browser based, which might mean that scanning IP addresses is the only option as bad as it is. Although there are maybe even more reasons that it is a bad idea because making it work would require opening up some cross origin request stuff which might be a security risk. 

Can you think of any browser based options for finding where the ESP32 is on the network which would be better?

When I do something like:
```
var xhr = new XMLHttpRequest();
xhr.open("POST", "http://192.168.0.133", true);
```

If there is an ESP32 at that address where in the firmware does that request land?

MitchBradley, 2021-06-30:
https://www.codeproject.com/Articles/1273691/SSDP-Discovery-in-JavaScript-and-Chrome

BarbourSmith, 2021-06-30:
I did read that and it's an option, but the fact that it is chrome only makes me a little worried. It feels a bit like the web is becoming Chrome exclusive and I'd like to avoid playing into that. 

I think id like to give port scanning a try if only to convince myself it's a bad idea.

Where in the firmware does a new incoming request land?

bdring, 2021-06-30:
Join our discord server

https://discord.gg/XDNH8WWbSc

MitchBradley, 2021-06-30:
How will the code get into the browser? It cannot be served from the ESP32 itself because that is what you are trying to find. If it is served from some external web site, you are in effect asking a user to run a web page that port scans the local network. The distinction between that and malware is, at best, razor thin.

luc-github, 2021-07-01:
Using SSDP or mDNS on local network will list devices but currently there is nothing implemented that allow to identify easily the GRBL or ESP3D based devices specificaly from other devices, it is in roadmap to use extra MDNS service but there was a bug in multi service esp32 core when I tested it and so I postponed the implementation: https://github.com/luc-github/ESP3D-WEBUI/issues/85

Using a wan address to scan local network is not correct approach - it will be blocked by firewall in best case and considered as an attack in worst case.

The hostname is suposed to avoid the need of IP address:
[image]

if you do not remember the name of your esp because you changed it, then under windows OS  it apear in other devices in network but you still need to check among the devices listed.
[image]

on Linux and OSX and android, there are no such feature, but they have some mdsn scanners, unfortunatly I do no know any which is cross platform.

If you want an easy solution, cross operating system, cross platform, you need to develop something like the ip cam search tools : https://wiki.instar.com/en/Quick_Installation/Alternative_IP_Scanner/ that will be run by user on his local network but not from internet.

BarbourSmith, 2021-07-01:
Thanks for the discord invite, I'm going to close this issue and move the conversation there which seems like a more appropriate place.

I agree with everyone who says that port scanning is ugly and bad and shares a lot of the same outward characteristics as malware....BUT at the same time I think one of the really fantastic and beautiful things about the GRBL ESP32 project is that it lets any machine which can connect to the wifi and has a browser control the machine. People can use an old android phone or a old computer if they have one laying around...but without an easy way to find the machine which is equally available on many platforms that ability is limited. For a lot of us reading the IP address off the serial port is trivial, but I think that there are even more people out there who don't want to learn about computers to build cool things out of wood who would benefit from a simpler connection method...if it's even possible.

closed 2021-07-01 by BarbourSmith

MitchBradley, 2021-07-01:
The service discovery problem is not specific to Grbl_Esp32.