bdring/Grbl_Esp32 #930: Small moves are not working

phorton1, 2021-07-02:
Line 338 of Planner.cpp needs to be changed to something like:

delta_mm = ((float)(target_steps[idx] - position_steps[idx])) / ((float)axisSettings[idx]->_stepsPerMm);

The problem was very reproducible in the tip of the Yaml_Settings branch at the time of this writing.

Small moves (less than 1 mm) were not working in either G0 or G1 modes.   The change of stepsPerMm from a float to an integer should probably be examined systemically.   The above assignment was resulting in delta_mm being assigned to 0.0000 for moves less than 1 mm causing "wonky" behavior (in my cases the steppers just started moving slowly in a positive direction and did not stop).   

Steps to reproduce:

G0 x200
G0 x200.1

I would submit a "push" request but I'm not sure exactly how to do that.

I have another small obvious bug as a test case for when I learn how to do push requests.

Have a nice day!

MitchBradley, 2021-07-02:
Thanks for the diagnosis, we will look into it.

MitchBradley, 2021-07-02:
Does it work with just
```
delta_mm = (float)(target_steps[idx] - position_steps[idx]) / axisSettings[idx]->_stepsPerMm;
```

MitchBradley, 2021-07-02:
Oh, hold on, are you using the YamlSettings branch?

MitchBradley, 2021-07-02:
The use of integers for steps_per_mm is a mistake.  I am fixing it.

MitchBradley, 2021-07-03:
Fixed

bdring, 2021-07-03:
I think all YamlSettings branch issues should be discussed on Discord.

It will cause confusion here.

DM me on Discord for an invitation to the channel

phorton1, 2021-07-03:
Really?   So I have signed up with a proprietary web service, for your benefit, who will sell my email address and track me .. but I dont know how to "DM" you or really want to learn their UI and ecosystem  ... "Wumpus has nobody to play with" ... sheesh.    Sigh.  Good morning ....

bdring, 2021-07-03:
I am not sure what you are trying to say.

Here is a direct link to our server.

https://discord.gg/XDNH8WWbSc

closed 2021-07-12 by bdring