bdring/Grbl_Esp32 #931: Single character typo on line 60 of Motors/RCServo.cpp

phorton1, 2021-07-02:
I dunno.  It's probably quicker for you for me to just tell you that the first %d on line 60 of RCServo.cpp should be changed to %s to match the (const char *) return type of _pwm_pin.name().c_str()

RCServo.cpp line 60:
-        info_serial("%s RC Servo Pin:%d Pulse Len(%.0f,%.0f) %s",
+        info_serial("%s RC Servo Pin:%s Pulse Len(%.0f,%.0f) %s",

Sometime, though, I'd like to understand how to properly make a push request.   I hope you got an email from me about that  Not sure how Github emali works or if you did.  Thanks for gargantuan efforts!

phorton1, 2021-07-02:
I should have added "low priority" ... not a critical, nor even a functional issue.

MitchBradley, 2021-07-02:
I made the change and will push it soon.

closed 2021-07-02 by MitchBradley

MitchBradley, 2021-07-03:
Fixed