bdring/Grbl_Esp32 #936: the laser does not turn off

benek1, 2021-07-04:
In fact, everything works, but the blue laser does not turn off during the cooling pause. the same when paused, there is also a peak when turned on, which causes a hole to burn.
GRBL_VERSION "1.3a";
GRBL_VERSION_BUILD = "20210424"
I tried other predefined machines. The problem is everywhere.
Working with this bug is very dangerous.
TXS0101 appears in the esp32_cnc_test_v4.1_schm.pdf diagram. his "OE" is connected to Sp_En but on this pin the signal changes in the same way as on the laser pin and also occurs during a break in traffic so it cannot be used. I am asking for any suggestions on how to solve this problem.
Setting up my machine
[ZDV_LASER.zip](https://github.com/bdring/Grbl_Esp32/files/6760041/ZDV_LASER.zip
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_ZDV_LASER]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:Global stepper disable pin:GPIO(13)]
[MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(14) Disable:None Limits(-210.000,0.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(26) Dir:GPIO(15) Disable:None Limits(-297.000,0.000)]
[MSG:PWM spindle Output:GPIO(2), Enbl:GPIO(22), Dir:GPIO(21), Freq:5000Hz, Res:13bits]

[MSG:Local access point GRBL_ESP started, 192.168.4.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:X  Axis limit switch on pin GPIO(5)]
[MSG:Y  Axis limit switch on pin GPIO(4)]
[MSG:Z  Axis limit switch on pin GPIO(18)]

Grbl 1.3a ['$' for help]
$Sta/SSID=Orange_Swiatlowod_B698
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=GRBL_ESP
$AP/Password=******
$AP/IP=192.168.2.1
$AP/Channel=1
$System/Hostname=grblesp
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=btgrblesp
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=10
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=Off
$Probe/Invert=Off
$Report/Status=1
$GCode/JunctionDeviation=0.001
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=Off
$Homing/DirInvert=
$Homing/Squared=
$Homing/Feed=100.000
$Homing/Seek=800.000
$Homing/Debounce=250.000
$Homing/Pulloff=2.000
$GCode/MaxS=1000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=On
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=PWM
$X/StepsPerMm=160.000
$Y/StepsPerMm=160.000
$Z/StepsPerMm=160.000
$A/StepsPerMm=8.000
$B/StepsPerMm=8.000
$C/StepsPerMm=8.000
$X/MaxRate=2000.000
$Y/MaxRate=2000.000
$Z/MaxRate=500.000
$A/MaxRate=80.000
$B/MaxRate=80.000
$C/MaxRate=80.000
$X/Acceleration=200.000
$Y/Acceleration=200.000
$Z/Acceleration=10.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=210.000
$Y/MaxTravel=297.000
$Z/MaxTravel=200.000
$A/MaxTravel=0.000
$B/MaxTravel=0.000
$C/MaxTravel=0.000
$X/Current/Run=0.250
$Y/Current/Run=0.250
$Z/Current/Run=0.250
$A/Current/Run=0.250
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.125
$Y/Current/Hold=0.125
$Z/Current/Hold=0.125
$A/Current/Hold=0.125
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=16
$Y/Microsteps=16
$Z/Microsteps=16
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
ok
ok

bdring, 2021-07-04:
This is due to the fact that the laser will turn on when it does not receive a signal. That is a pretty dangerous situation, but I think they do it that way so you can turn the laser on by the power only. This is not a bug in the firmware. 

There is no perfect solution to this. Before turn on and after turn off, the laser can still fire. The best solution is shown in this machine definition. It simply turns on the enable full time at turn on and keeps it on regardless of gcode, etc.

https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Machines/mpcnc_laser_module_v1p2.h

See the line 33 and 34

benek1, 2021-07-04:
there is no pin 16 in my ESP. I mapped to GPIO2. and no signal. it still does not work.
more dangerous is that the laser is on pause

bdring, 2021-07-04:
Yes, I understand you have a dangerous laser.

I am on vacation this weekend away from my computer, but I can try to help you out a little.

Try this machine definition.
[3axis_lasel_4p1.zip](https://github.com/bdring/Grbl_Esp32/files/6760460/3axis_lasel_4p1.zip)

Keep in mind this will not fix the laser when the controller is not on.

richardwpaul, 2021-07-04:
I hope I am  posting this question correctly. I have two machines that are already using the 6pack. I want to build a machine similar to the RatRig V-core 3. I think GRBL-ESP32 would be perfect for this. But I need to independently control 3 motors for the Z axis. Any chance of this happening in the near future? Rick Paul765.430.5389*HP* From: bdringSent: Sunday, July 4, 2021 3:07 PMTo: bdring/Grbl_Esp32Cc: SubscribedSubject: Re: [bdring/Grbl_Esp32] the laser does not turn off (#936) Yes, I understand you have a dangerous laser.I am on vacation this weekend away from my computer, but I can try to help you out a little.Try this machine definition.3axis_lasel_4p1.zipKeep in mind this will not fix the laser when the controller is not on.

ggallant571, 2021-07-05:
I put a 3.3K pull-down between the PWM and GND.

MitchBradley, 2021-07-05:
That usually helps a lot.  In one instance, I think I had to use an even lower-resistance pull-down to ensure that it drove the signal below the input threshold. You could measure the voltage there with nothing else driving it.  Make sure the voltage is well below 1.7V to be safe.
I usually put the resistor right on the laser module, to guard against the laser turning on if the PWM cable comes loose.

ggallant571, 2021-07-05:
Until the I/O pin is defined as an output and driven low it will be will 
float high. Solution is to force it low. Connect a 3.3K (approximate) 
resistor between the PWM signal and ground. Cost is about $0.02US.

benek1, 2021-07-05:
Unfortunately. I loaded the program with the 3axis_lasel_4p1 settings and everything the old way. During laser pause and cooling, the laser is turned on.
Maybe someone smarter than me will improve the program so that it can be safely used with a diode laser. The sea in a while. Now I am sure that it is not my settings and not my PCB is the reason (I will wait for the delivery of TXS0101, I doubt that it will change something) Maybe some good person will come up with something else?

benek1, 2021-07-05:
The software behaves correctly with respect to grbl 0.9 and lower, only in version 1.1 the laser is turned on only while it is moving. Is it possible to implement functions from grbl 1.1 in the current version of ESP32

benek1, 2021-07-05:
Thank you for all your help and I close this thread.
At the same time, I open a request for a new function similar to grbl 1.1
Thanks again

closed 2021-07-05 by benek1