bdring/Grbl_Esp32 #937: Laser Grbl

benek1, 2021-07-05:
I noticed that the current version of GRBL works like version 0.9, which means that it is perfect for spindle plotters.
I am a diode laser user and I miss the features from version 1.1 - the laser is only turned on while it is moving. this enables automatic laser cooling and pausing. I am convinced that there are many owners of such lasers and many of them would like to switch to the 32bit version.

bdring, 2021-07-05:
We already support that feature. It works just like Grbl 1.1. You must turn on laser mode via.

$Gcode/LaserMode=On

or 

The $ number setting Grbl uses that I can no longer remember.

benek1, 2021-07-06:
It does not work. The laser is on when it is not moving. $ 32 = 1 laser mode is set.
  defining SPINDLE_DIR_PIN enabled M4 support, but during laser pause and automatic laser cooling, the laser remains on. the SPINDLE_ENABLE_PIN pin is high, i.e. on, the SPINDLE_DIR_PIN pin is high and does not change.
Looking at the Grbl_ESP32_Development_Controller diagram, I can see a voltage level translator. His OE pin is connected to SPINDLE_ENABLE_PIN, it looks like he is watching the laser status,
however the status on this pin does not change. It doesn't work anyway.
Maybe you have any suggestions?

benek1, 2021-07-08:
what the pin is for LVL_SHIFT_ENABLE

benek1, 2021-07-08:
Thanks, I made it.

closed 2021-07-08 by benek1