bdring/Grbl_Esp32 #938: X-Y Axis Travel Too Short, Z-Axis Fine

kevin0730, 2021-07-07:
I am just getting my PrintNC machine running and struggled all day to figure out why the X and Y axes do not travel the commanded distance but Z is fine.  I have 10mm ball screws on X-Y and a 4mm ball screw on Z.  I have the same Nema 23 1.8 degree motors on all axes.

If I command 10mm on X or Y I get 6.7mm movement.  If I command 100mm I get about 96mm.  I have tried all sensible combinations of micro stepping on my DM542T drivers to no avail.  I did a no-load test, commanding the stepper to give 10mm travel and it did not rotate a full revolution, about 70% like I see in my dial gauge measurements.

Bart suggested increasing all of the pulse delay times but that did not change the behavior.

joedirium, 2021-07-07:
How many micro steps per rotation did you configure on your stepper driver?
Which resolution did you configure in grbl_esp32 for the same axis ($100 for the x-Axis)?
Acceleration values are reasonable ($120 for x-Axis)? No steps lost when ramping up?
Can you connect a logic analyzer to the step wire close to the stepper and measure the number of pulses to check if it is a software/electrical problem or a stepper/mechanical problem?

kevin0730, 2021-07-07:
I have tried several microstepping settings, right now it is set to 40 steps/mm and the DM542T is set to 2.  All microstepping combinations I have tried have the same error.

 

I have a USB-type logic analyzer that I can connect to 5V logic, at the stepper it will by my supply voltage (correct?) which is 36V.

 

Kevin

joedirium, 2021-07-07:
To avoid misunderstanding: The DM542T allows 400-25000 steps per rev to be set by dip switches SW5 to SW8.
How did you set these switches?

The step signal comes from your ESP32 board. The power supply is for the output to the motors. 
Right, I did not ask you accurately where to do the measurement. It should be done at the step input of the DM542, which is s closest possible point to the motor, but still far away.

kevin0730, 2021-07-07:
I now have the system set to 200 steps/mm with switches SW5 to SW8 set OFF ON ON OFF

 

On advise from the Discord forum I think we are on to the problem.  The acceleration was set to 200mm/sec^2, with it set to a tenth of that it is much more accurate.  I will be spending some more time making parameter adjustments to see what improvements I can make.

 

The need to use the logic analyzer may be overcome by events.

joedirium, 2021-07-07:
0110 should be 2,000 steps per rotation. (I personally use 1600 at max. More steps means less power means higher potential for step loss.)
With a 10mm ballscrew (I assume you mean 10mm movement per rotation) it results in a resolution of 200 steps per mm as you said.
200mm/sec^2 is pretty much, even for nema23. I am using 50mm/sec^2 with a 5mm/rotation spindle. So 20mm/sec^2  should be a good choice to start with a 10mm/rotation spindle.

I assume you have more mechanics/configuration relation problem, so no need to use the logic analyzer. 

Anyway, the result should finally be highly accurate, not just "more accurate".

kevin0730, 2021-07-07:
I think I am going to reduce to 1,000 steps per revolution so I get 100 steps/mm.

 

Once I restore all of the things I disturbed in my troubleshooting I will get back to determining the accuracy which I should expect to be very precise.

kevin0730, 2021-07-10:
I have the controllers set to 1000 steps/rev and the acceleration at 10mm/s^2 on each axis and the travel is accurate.  I think we can say that this problem is closed.

closed 2021-07-10 by bdring