bdring/Grbl_Esp32 #939: Telnet to serial win utility?

whoim2, 2021-07-08:
Anybody use this way? Need free soft for create virtual serial port, and connect this to esp32grbl telnet.

HuubBuis, 2021-07-08:
You could use a second ESP32 to connect to the first one and create a bridge between the ESP's
There a a lot of examples like this one: [WiFi bridge](https://awesomeopensource.com/project/AlphaLima/ESP32-Serial-Bridge)

whoim2, 2021-07-08:
> You could use a second ESP32, USB connected to the PC and WiFi to connect to the first one, and create a bridge between the ESP's
> There a a lot of examples like this one: [WiFi bridge](https://awesomeopensource.com/project/AlphaLima/ESP32-Serial-Bridge)

Im try https://www.hw-group.com/software/hw-vsp3-virtual-serial-port program, its work

closed 2021-07-08 by whoim2

HuubBuis, 2021-07-08:
>Im try https://www.hw-group.com/software/hw-vsp3-virtual-serial-port program, its work

 I didn't know this program. I will give it a try some day. It could be useful for connecting gcode senders that don't support telnet.

Thanks

whoim2, 2021-07-08:
> > Im try https://www.hw-group.com/software/hw-vsp3-virtual-serial-port program, its work
> 
> I didn't know this program. I will give it a try some day. It could be useful for connecting gcode senders that don't support telnet.
> 
> Thanks

Yes, this work for senders with only serial connections