bdring/Grbl_Esp32 #940: Huanyang VFD 3.3v TTL to RS485

MisterDiz, 2021-07-12:
I have a standard Huanyang HY 1.5kHz VFD and am trying to set up spindle control with RS485.

I am assuming that the settings on the VFD are all good because the Spindle starts and spins as expected when hooked up directly to a PC using the wonderful Spindle Talker application. Unfortunately, I am getting no joy from my GRBL ESP32.

I’ve spent a few days on this and have basically got to the stage of giving up/becoming violent with it or myself.
I would be very appreciative for any help to confirm that what I’m doing hasn’t some cringingly basic errors and may be of use to others.

The only configuration I have made is to mymachine.h profile:

#define SPINDLE_TYPE SpindleType::HUANYANG
#define HUANYANG_TXD_PIN        GPIO_NUM_32
#define SPINDLE_RXD_PIN         GPIO_NUM_33

Is that all I need to do code wise?

I’m using a 3.3v compatible TTL to RS 485 board from Amazon that looks like this:

￼
[image]

I’ve connected the TX, RX, 3.3v & GND pins to the corresponding ones on the ESP32 with A+, B- to RS+, RS-. From what I can glean, it seems that the consensus is Ground on the output side is not connected to the VFD. (Also tried swapping TX and RX).

In UGS Platform (which is working for motor control, coolant relay etc), I’m sending an M3 S1000 command to the ESP32 but am not getting any flashing TX lights or action. I don’t have a scope to see what’s going on but the voltage on the TTL to RS485 board across the 3.3V/GND input is present and across the A+ B- is showing 4.8ish volts.

_As an aside, when I set the VFD up to accept external voltage control rather than RS485, I can use the 0-5v input setting on the VFD to read a 0-3.3v PWM signal, (just to see if anything works). M3 S1000 changes the frequency setting on the VFD but doesn’t start the spindle (not sure why or if this is related)._

Is there anything I’m missing? Any settings that I haven’t made? Is sending “M3 S1000” enough to run the spindle? 

Apologies if there are one or more dumb noob moves going on but I’ve done a fair bit of trawling and can’t seem to get anywhere. It would seem that I'm trying the most commonly used parts and it may be of use to others to have a neat way to get a working RS485 HY spindle setup. Any help is gratefully received.

bdring, 2021-07-12:
Can you paste the [boot messages](https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages) here?

Can you confirm your VFD setting are set like this...

    PD004   400  Base frequency as rated on my spindle (default was 50)
    PD005   400  Maximum frequency Hz (Typical for spindles)
    PD011   120  Min Speed (Recommend Aircooled=120 Water=100)
    PD014   10   Acceleration time (Test to optimize)
    PD015   10   Deceleration time (Test to optimize)
    PD023   1    Reverse run enabled
    PD141   220  Spindle max rated voltage
    PD142   3.7  Max current Amps (0.8kw=3.7 1.5kw=7.0, 2.2kw=??)
    PD143   2    Poles most are 2 (I think this is only used for RPM calc from Hz)
    PD144   3000 Max rated motor revolution at 50 Hz => 24000@400Hz = 3000@50HZ

    PD001   2    RS485 Control of run commands
    PD002   2    RS485 Control of operating frequency
    PD163   1    RS485 Address: 1 (Typical. OK to change...see below)
    PD164   1    RS485 Baud rate: 9600 (Typical. OK to change...see below)
    PD165   3    RS485 Mode: RTU, 8N1

MisterDiz, 2021-07-12:
Thanks for the amazingly quick reply.

The VFD settings are as you describe, I took them from HuanyangSpindle.cpp
SpindleTalker reads them back as:

[image]

The boot message spills the beans that the pins are undefined but I have set them up as per my initial post.
I am editing VSCode but uploading from Arduino IDE using the "local" sketch folder

_[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 4]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:X  Axis Standard Stepper Step:GPIO(4) Dir:GPIO(16) Disable:None Limits(0.000,300.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(17) Dir:GPIO(5) Disable:None Limits(0.000,300.000)]
[MSG:Z  Axis Standard Stepper Step:GPIO(18) Dir:GPIO(19) Disable:None Limits(-300.000,0.000)]_
**[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_TXD_PIN]
[MSG:Undefined VFD_RS485_RXD_PIN]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:RS485 VFD spindle errors]**

_[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin GPIO(13)]
[MSG:X  Axis limit switch on pin GPIO(21)]
[MSG:Y  Axis limit switch on pin GPIO(22)]
[MSG:Z  Axis limit switch on pin GPIO(23)]
[MSG:A  Axis limit switch on pin GPIO(27)]
[MSG:Probe on pin GPIO(14)]_

MitchBradley, 2021-07-12:
Since you are already using VsCode, you could install platformio and use that instead of Arduino.  Platformio does everything that Arduino IDE does, only better and faster. All of use developers use platformio, and most of us use it under VsCode.

MitchBradley, 2021-07-12:
https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-PlatformIO

MisterDiz, 2021-07-12:
Thanks for the heads up. I just reinstalled platformio and reuploaded with the same outcome. It definitely seemed quicker to upload though 👍

bdring, 2021-07-12:
The Wiki has a few examples.

https://github.com/bdring/6-Pack_CNC_Controller/wiki/RS485-Modbus-Module

jelsdon, 2021-07-12:
I'm using the following definitions with success against similar hardware

```
#define SPINDLE_TYPE SpindleType::HUANYANG
#define VFD_RS485_TXD_PIN GPIO_NUM_4
#define VFD_RS485_RTS_PIN GPIO_NUM_2
#define VFD_RS485_RXD_PIN GPIO_NUM_27
```

MisterDiz, 2021-07-12:
Woohoo!

Changed:
#define HUANYANG_TXD_PIN          GPIO_NUM_32
#define SPINDLE_RXD_PIN          GPIO_NUM_33

to:
#define VFD_RS485_TXD_PIN        GPIO_NUM_32
#define VFD_RS485_RXD_PIN        GPIO_NUM_33

and added:
#define VFD_RS485_RTS_PIN        GPIO_NUM_25

That cleared the error, and I have the joy of flashing RS485 lights and a fully functioning spindle, I can send correct S24000 messages and spin in either direction. The little board seems like a cheap and easy way to get an HY VFD working with the ESP32 3.3v outputs.

Thanks very much for helping find a solution, I hope someone else finds it useful.

One last thing, I had left the RTS pin undefined because my (seemingly incorrect) understanding was that it was optional. But defining it was a necessary step to get everything to work, however the pin is not physically connected to anything. My hope was that I wouldn't have to use up 3 precious pins. Is that incorrect?

MisterDiz, 2021-07-12:
If I have to assign 3 pins, could I use one of the GPI pins 34, 35, 36, 39 for RXD? Presumably GPIO 2 is usable for TXD?

jelsdon, 2021-07-13:
Apologies, I'd managed to delete my previous post - looks like you have it sorted though

>  I'm using the following definitions with success against similar hardware
> 
> ```
> #define SPINDLE_TYPE SpindleType::HUANYANG
> #define VFD_RS485_TXD_PIN GPIO_NUM_4
> #define VFD_RS485_RTS_PIN GPIO_NUM_2
> #define VFD_RS485_RXD_PIN GPIO_NUM_27
> ```

My setup does differ as I'm using a max485/rs485 module with `di` `de` `re` `ro` pins for the esp32 side, `de` and `re` connected to RTS

MisterDiz, 2021-07-13:
Thanks jelsdon :)

Its working which is great, pin 2 -> TXD is good, pin 15 also works for RXD. All my 'safe' pins are loaded, so it would be great if anyone knew if I could 'hide' the pin for RTS that doesn't need physically attaching to anything (pin 12?) or if I could use a GPI pin for RXD?

MitchBradley, 2021-07-13:
I think, based on a quick look through the driver code both in Grbl_Esp32 and in the underlying software development kit, that it would be possible to leave VFD_RS485_RTS_PIN undefined - except for the error check that prevents it from working.  In other words, the only thing that would prevent it from working is the error check itself.  The code is in Grbl_Esp32/src/Spindles/VFDSpindle.cpp.  It looks like this:
```
        if (_rts_pin == -1) {
            grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Info, "Undefined VFD_RS485_RTS_PIN");
            pins_settings_ok = false;
        }
```
I think it would work if you comment that out by adding #if 0 .. #endif, like this:
```
#if 0
        if (_rts_pin == -1) {
            grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Info, "Undefined VFD_RS485_RTS_PIN");
            pins_settings_ok = false;
        }
#endif
```

MitchBradley, 2021-07-14:
We are working on the next generation code which will let you assign pins without recompiling.  I will add the ability to leave the rts pin unassigned to that version.

MitchBradley, 2021-07-14:
I created a ticket in our next generation work list to do this.

closed 2021-07-14 by MitchBradley