bdring/Grbl_Esp32 #942: Changing Radio mode with [ESP115] command

ithinkido, 2021-07-12:
I am not able to change the Radio State from BT to AP or STA using [ESP115] STA or [ESP115] AP as described in the wiki
[image]
What version of the firmware are you using?
1.3a

Is the problem repeatable?
yes

**To Reproduce**
1. from the webui on a stock testdrive GRBL ESP32 change the radio mode to BT
2. Open a serial connection to the borad
3. Send [ESP115] STA or [ESP115] AP
4. Reply is "only ON or OFF mode supported!"

[image]

bdring, 2021-07-12:
It appears to be working via the WebUI ESP3D tab, but not via the WebUI or serial console.

setRadioState(...) is not checking those other options.

ithinkido, 2021-07-13:
How are you accessing the WebUI when in BT node ?

bdring, 2021-07-14:
The WebUI works in a web browser, so it needs WiFi.

No Bluetooth

ithinkido, 2021-07-14:
Yes, and that is the issue. When in BT mode , it is not possible to change the radio mode with [ESP115] commands. Hence my curriosty to your comment, "It appears to be working via the WebUI ESP3D tab,."

MitchBradley, 2021-07-14:
The wiki documentation is wrong.  I just fixed it. ESP115 does not switch between different kinds of radios - it turns the radio on or off.  ESP115 is not persistent.  After you reboot, its effect is lost.  ESP110, also known as $Radio/Mode is the command to use to select the radio.  [ESP110]STA  or preferably $Radio/Mode=STA

closed 2021-07-14 by MitchBradley

ithinkido, 2021-07-15:
The $ESP command still has the wrong info
[image]

MitchBradley, 2021-07-17:
Thanks.  I just fixed that problem in the YamlSettings branch that will become the base for the new code.  Fixing it in Devt is probably not worthwhile right now because we are working full-time on the new branch.