bdring/Grbl_Esp32 #943: Wifi is getting stuck on the loading screen

Protoneer, 2021-07-15:
**What version of the firmware are you using?**
Latest version
[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.3.4-432-g7a85334d8]

**Is the problem repeatable?**
Yes - Tried it on the 6-pack-CNC esp32 and my own ESP32 modules.

**Under what conditions does the bug occur?**
I was able to load the website GZ and after that I was not able to load the website again.
Console is working as normal...

```
[MSG:Connected with 10.82.0.173]
[MSG:Start mDNS with hostname:http://grblesp.local/]

```

MitchBradley, 2021-07-15:
Which browser?  If you are using Chrome, try opening Developer Tools to see if the console shows any error messages.

MitchBradley, 2021-07-15:
Also, what is your configuration?  Which Machine.h file, and any changes to Config.h or Defaults.h.

Protoneer, 2021-07-17:
> Which browser? If you are using Chrome, try opening Developer Tools to see if the console shows any error messages.

No errors....

Tried it with test_drive.h

I did notice even after re programming with Arduino it remembered my network settings.

So I have erased the ESP32 with:
```
esptool --chip esp32 --port COM5 erase_flash
```

Seems to work on a spare ESP32. Will try it on my test machine on Monday.

Protoneer, 2021-07-19:
```
esptool --chip esp32 --port COM5 erase_flash
```

Did the trick....  ;)

closed 2021-07-19 by Protoneer