bdring/Grbl_Esp32 #946: M4 commands not accepted in Laser Mode with PWM spindle

Spark-Concepts, 2021-07-16:
Please answer the following questions.

What version of the firmware are you using? 
- Ver 1.3a Date 20210424

Is the problem repeatable?
 - Yes 

Under what conditions does the bug occur?
- when using laser mode, $32=1, M4 commands are not accepted - kick backs error 20.   Tested with a fresh install of all libraries and directories.  Using the SPI_DAISY_4X_XYZ machine definition with no changes.  Startup message, and error below.  

any thoughts?  

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:SPI_DAISY_4X_XYZ]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC2130 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.110 Limits(0.000,300.000)]
[MSG:X  Axis Trinamic driver test failed. Check motor power]
[MSG:Y  Axis Trinamic TMC2130 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.110 Limits(0.000,300.000)]
[MSG:Y  Axis Trinamic driver test failed. Check motor power]
[MSG:Z  Axis Trinamic TMC2130 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:3 R:0.110 Limits(-300.000,0.000)]
[MSG:Z  Axis Trinamic driver test failed. Check motor power]
[MSG:PWM spindle Output:GPIO(25), Enbl:GPIO(4), Dir:None, Freq:5000Hz, Res:13bits]
[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(36)]
[MSG:Y  Axis limit switch on pin GPIO(39)]
[MSG:Z  Axis limit switch on pin GPIO(34)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
> $32=1
ok
> M4 S100
[MSG:M4 requires laser mode or a reversable spindle]
error:20 (Unsupported command)

MitchBradley, 2021-07-16:
SPI_DAISY_4X_XYZ has 
#define SPINDLE_TYPE            SpindleType::RELAY

Relay spindles do not support Laser mode and are not otherwise reversable.

MitchBradley, 2021-07-16:
The "with PWM Spindle" in the ticket name is inconsistent with " Using the SPI_DAISY_4X_XYZ machine definition with no changes."

Spark-Concepts, 2021-07-16:
I used this machine file here: https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Machines/spi_daisy_4axis_xyz.h

PWM defined line 68, with GPIO 25 (output) and GPIO 4 (enable).  

I am noticing the machine name, and the .h file name do not match.

MitchBradley, 2021-07-16:
For M4 to work with PWM, you must define a direction pin, otherwise there is no way to go counterclockwise.  If you want to use laser mode, you have to use SpindleType::Laser

Spark-Concepts, 2021-07-17:
Ok - quick sanity check for me please.  For Laser mode to work, we need to define a new machine file that has:

SpindleType::Laser
LASER_OUTPUT_PIN
LASER_ENABLE _PIN  
$Gcode/LaserMode=on

This correct?

bdring, 2021-07-17:
LASER_ENABLE_PIN is optional and only used if your laser requires it.

Spark-Concepts, 2021-07-17:
Copy on the Laser_Enable_pin :).  

The main question is:  does turning on LaserMode (real time setting) allow you to go from PWM driven Spindle to a PWM driven Laser that accepts M3 and M4 commands from Lightburn?

From the above discussion, I think we need to define a new machine file for the PWM driven Laser and push a firmware update when we swap from the pwm spindle to pwm laser - is this correct?

bdring, 2021-07-17:
You can switch from PWM to Laser at run time if you define both in your machine definition. Then you just send $Spindle/Type=Laser or $Spindle/Type=PWM to switch types. They can use the same pins or separate pins.

https://github.com/bdring/Grbl_Esp32/wiki/Spindle-Types#overview

MitchBradley, 2021-07-17:
The new code that we are working on, very nearly ready for testing, lets you setup multiple spindles in a single machine configuration.  Each spindle has a range of tool numbers, for example the rotating spindle might be tool numbers 0..9 and the laser would be tool number 10 and above.  When you issue a GCode M6 Tn command, the spindle for n is activated.

In the old code, you can define both SPINDLE_OUTPUT_PIN and LASER_OUTPUT_PIN in the same machine file.  You can switch from spindle to spindle by issuing $Spindle/Type=Laser or $Spindle/Type=PWM .  To enable or disable laser mode, you issue $laser/mode=on (or off).  If you always want laser mode when using the laser spindle, you can just leave $laser/mode=on, because when the PWM (rotating) spindle is selected, Rlaser/mode is ignored.

Spark-Concepts, 2021-07-18:
Thanks Gents!  I was unaware the you could assign the GPIOs to multiple spindle functions.  Made the changes and it works perfectly!

closed 2021-07-18 by Spark-Concepts

bdring, 2021-07-18:
Dual function, laser/spindle machine are popular these days.