bdring/Grbl_Esp32 #949: sdcard not work

whoim2, 2021-08-07:
Hello. Used this shematic: https://github.com/bdring/Grbl_ESP32_Development_Controller
custom pcb: https://github.com/whoim2/esp32-grbl-3axis
cnc work normally, but im try use sd card - connection error at "refresh" button on interface
Esp32 flash blue led 5-10 seconds. Im use 2gb microSD and format it to fat16 with utinity from wiki.
Im need pullup resistors maybe?

t-web007, 2021-08-08:
i have the same error. i thing the TrinamicDriver uses spi and the sd card too. i try to compile without the TrinamicDriver no luck.

whoim2, 2021-08-08:
> i have the same error. i thing the TrinamicDriver uses spi and the sd card too. i try to compile without the TrinamicDriver no luck.

Im use 4988, no other spi connections. Use 3axis_v4.h machine file, enable mist/flood and input pins, pwm spindel.
SPI pins connected only to SD holder.

Trying 2pcs sd cards, fat16/fat32 fs. Check all connections many times. No effect.
Tomottow im replace sd connector, may be, its broken..

whoim2, 2021-08-08:
im connect to usb/uart and see: board restarted on "refresh" sd button on web:
`23:48:31.962 -> ***ERROR*** A stack overflow in task clientCheckTask has been detected.
23:48:31.996 -> abort() was called at PC 0x40090040 on core 1
23:48:31.996 -> 
23:48:31.996 -> ELF file SHA256: 0000000000000000
23:48:31.996 -> 
23:48:31.996 -> Backtrace: 0x4008fdac:0x3ffd1750 0x40090029:0x3ffd1770 0x40090040:0x3ffd1790 0x400928e7:0x3ffd17b0 0x400944b0:0x3ffd17d0 0x40094466:0x00000001
23:48:31.996 -> 
23:48:31.996 -> Rebooting...
23:48:31.996 -> ets Jun  8 2016 00:22:57
23:48:31.996 -> 
23:48:31.996 -> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
23:48:31.996 -> configsip: 0, SPIWP:0xee
23:48:31.996 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
23:48:32.030 -> mode:DIO, clock div:1
23:48:32.030 -> load:0x3fff0018,len:4
23:48:32.030 -> load:0x3fff001c,len:1216
23:48:32.030 -> ho 0 tail 12 room 4
23:48:32.030 -> load:0x40078000,len:10944
23:48:32.030 -> load:0x40080400,len:6388
23:48:32.030 -> entry 0x400806b4
23:48:32.607 -> 
23:48:32.607 -> [MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
23:48:32.607 -> [MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
23:48:32.607 -> [MSG:Using machine:ESP32_V4]
23:48:32.607 -> [MSG:Axis count 3]
23:48:32.607 -> [MSG:RMT Steps]
23:48:32.607 -> [MSG:Door switch on pin GPIO(35)]
23:48:32.607 -> [MSG:Reset switch on pin GPIO(34)]
23:48:32.641 -> [MSG:Hold switch on pin GPIO(36)]
23:48:32.641 -> [MSG:Start switch on pin GPIO(39)]
23:48:32.641 -> [MSG:Init Motors]
23:48:32.641 -> [MSG:Global stepper disable pin:GPIO(13)]
23:48:32.641 -> [MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(14) Disable:None Limits(0.000,290.000)]
23:48:32.641 -> [MSG:Y  Axis Standard Stepper Step:GPIO(26) Dir:GPIO(15) Disable:None Limits(0.000,370.000)]
23:48:32.641 -> [MSG:Z  Axis Standard Stepper Step:GPIO(27) Dir:GPIO(33) Disable:None Limits(-46.000,0.000)]
23:48:32.674 -> [MSG:PWM spindle Output:GPIO(2), Enbl:None, Dir:None, Freq:5000Hz, Res:13bits]
23:48:32.776 -> 
23:48:32.776 -> [MSG:Client Started]
23:48:32.776 -> [MSG:Connecting wifi]
23:48:32.776 -> [MSG:Connecting.]
23:48:33.285 -> [MSG:Connecting..]
23:48:33.761 -> [MSG:Connecting...]
23:48:34.271 -> [MSG:Connecting....]
23:48:34.781 -> [MSG:Connecting.]
23:48:35.257 -> [MSG:Connecting..]
23:48:35.767 -> [MSG:Connecting...]
23:48:36.277 -> [MSG:Connecting....]
23:48:36.583 -> [MSG:Connected with 192.168.0.110]
23:48:36.787 -> [MSG:Start mDNS with hostname:http://grbl-cnc.local/]
23:48:36.787 -> [MSG:SSDP Started]
23:48:36.787 -> [MSG:HTTP Started]
23:48:36.787 -> [MSG:TELNET Started 23]
23:48:36.787 -> [MSG:X  Axis limit switch on pin GPIO(17)]
23:48:36.787 -> [MSG:Y  Axis limit switch on pin GPIO(4)]
23:48:36.821 -> [MSG:Z  Axis limit switch on pin GPIO(16)]
23:48:36.821 -> [MSG:Probe on pin GPIO(32)]
23:48:36.821 -> 
23:48:36.821 -> Grbl 1.3a ['$' for help]
23:48:36.821 -> [MSG:'$H'|'$X' to unlock]
`

t-web007, 2021-08-08:
i have an old version without the TrinamicDriver this works. i do not use this driver and i can't compile the firmware without this driver.

MitchBradley, 2021-08-08:
To fix the stack overflow, change line 122 of src/Serial.cpp from
```
                           4096,               // size of task stack
```
to
```
                            8192,               // size of task stack
```

whoim2, 2021-08-08:
> To fix the stack overflow, change line 122 of src/Serial.cpp from
> 
> ```
>                            4096,               // size of task stack
> ```
> 
> to
> 
> ```
>                             8192,               // size of task stack
> ```

Wow! Worked! Thanks!
[image]

whoim2, 2021-08-08:
i think, this need adding to a main code?

closed 2021-08-08 by whoim2