bdring/Grbl_Esp32 #950: Installing Firmware on ESP32 UNO layout board and CNC Shield

jjhamb, 2021-08-08:
Hi there..
I live in India and I do not have access to your ESP32 board. 
I have GRBL Plotter (AxiDraw Clone) running well on UNO with CNC Shield V3.
I am wondering if your firmware can be configured to run ESP32 with UNO layout + CNC Shield V3 and read gcode from SD Card.

My reason for switch are two:
1. Using SD Card
2. Wifi Access.

https://www.botnroll.com/en/esp/3639-wemos-d1-r32-w-esp32-uno-r3-pinout.html
[image]

https://www.amazon.in/Shield-A4988-DRV8825-Driver-Expansion/dp/B077TDDZ4V
[image]

bdring, 2021-08-08:
See this blog post

https://deeemm.com/cnc/2021/07/15/ESPduino-protoneer-CNC-Shield.html

jjhamb, 2021-08-08:
Hey thanks.....
Any pointers for SD card access?

bdring, 2021-08-08:
I don't know much about that. I just recalled previous issues and sent the link. 

Search the issues. Maybe you will find something.

jjhamb, 2021-08-08:
Thanks Noted..

closed 2021-08-14 by bdring

4klemo4, 2022-02-07:
I have set my espduino like the blog https://deeemm.com/cnc/2021/07/15/ESPduino-protoneer-CNC-Shield.html. Everything working except laser pwn. Laser is always on no matter what I do. I have try arduino uno and grbl 1.1 and is working fine. Has anyone successfully set up laser control on espduino?