bdring/Grbl_Esp32 #953: Independent homing limit switch position for dual Y axis squaring

allistar, 2021-08-12:
I have an XYYZ machine and I'd like to get axis squaring working. It's going to be very difficult to ensure the two Y limit switches are perfectly aligned, and even 0.5mm difference isn't good enough. To help with this it would be helpful if we can specify an independent offset for each limit switch. E.g. that the Y2 limit switch is 0.5mm offset from the Y1 limit switch.

I don't think axis specific pull off values will work as the home position doesn't include the pull off value, as seen here:
https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Limits.cpp#L246

Perhaps the code:
sys_position[idx] = (mpos + pulloff) * steps;

Could be changed to:
sys_position[idx] = (mpos + pulloff + limitOffset) * steps;

Where "limitOffset" is an axis specific offset set in config, or 0 if undefined.

This will make gantry sharing on XYYZ machines easier to configure.

I think this feature will appeal to users of XYYZ setups which are common.

Bakafish, 2021-08-12:
If you look in the YamlSettings& FluidHoming branches you will see that as part of the configuration changes, they are also re-architecting the limits and axis homing functionality with your request being addressed.

allistar, 2021-08-12:
Thanks, I'll take a look.

Bakafish, 2021-08-12:
Keep in mind, these branches are still in development and are not ready for Beta testing yet, but you can at least confirm the functionality is there.

allistar, 2021-08-12:
My controller (XPro V5) has dual Y limit switch inputs, so I wouldn't need to "rack" the gantry to home, the fluid homing sounds a lot better. I look forward to this being tested, released and then pulled downstream.

bdring, 2021-08-12:
If the switches are not square it must rack to detect them then go back to square with the pull off.

closed 2021-08-14 by bdring