bdring/Grbl_Esp32 #954: what is the rate per/second if i use all the 6axis ?

bd7jat, 2021-08-13:
As the topic said, Is there anybody know the max rate when using 6 axis at the same time ?Thanks a lot

MitchBradley, 2021-08-13:
The question is incomplete.  The 6 pack board can step at upwards of 100kHz on all axes simultaneously, but the rate that corresponds to depends on motor parameters, axis gearing, axis geometry, and other factors.

closed 2021-08-13 by MitchBradley

Bakafish, 2021-08-13:
The six-pack uses shift registers linked to the i2s interface. I don't think the number of axis in use has any effect on the maximum refresh rate in such an implementation. That rate, in your specific application, is also going to have a lot of external dependencies like the signal path and steppers used.