bdring/Grbl_Esp32 #955: Problems Compiling Firmware: Arduino IDE v PlatformIO

Project487, 2021-08-16:
Hi,

I have compiled Grbl_Esp32 successfully using the Arduino IDE and PlatformIO.  The Arduino version works on my Esp32_Dev board but the PlatformIO version does not.  I'm using the latest versions of everything and using unmodified Grbl_Esp32 files from the same directory.  Both Windows10 64bit and Linux Mint 20 tried.  The PlatformIO compilation gives no errors and uploads successfully.

These are the results form the serial monitor (PuTTY):

Arduino compiled:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:Using machine:Test Drive - Demo Only No I/O!]
[MSG:Axis count 3]
[MSG:Timed Steps]
[MSG:Init Motors]
[MSG:No spindle]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.3a ['$' for help]
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>`

PlatformIO compiled:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8`

I want to compile a different machine version (rather than 'test_drive') and require PlatformIO to work.  Any help appreciated.

Thanks

Roger

======================= 

Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?
Yes
What version of the Arduino IDE are you using? 
1.8.13
What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 
12 Aug 2021
Are you using the master branch of Grbl_ESP32?
Yes
Have you made any edits or configuration changes (list them) to the  firmware?
No
Please paste the compiler error text here:
No errors from compiler

Bakafish, 2021-08-16:
What branch are you using? The latest version of PlatformIO seems to require you to change the setting of `board_build.flash_mode` to "`board_build.flash_mode = dio`" in the `platformio.ini` file on the top level.

Project487, 2021-08-16:
> 
> 
> What branch are you using? The latest version of PlatformIO seems to require you to change the setting of `board_build.flash_mode` to "`board_build.flash_mode = dio`" in the `platformio.ini` file on the top level.

Thanks for the reply; I'll investigate the settings.

Master branch of Grbl_ESP32
PlatformIO: Core 5.1.1  Home 3.3.4

Project487, 2021-08-17:
Further update.

Despite a 'successful' build

```
Building in release mode
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.0% (used 13224 bytes from 327680 bytes)
Flash: [==        ]  15.4% (used 201212 bytes from 1310720 bytes)
========================= [SUCCESS] Took 8.14 seconds =========================

```

it appears there are some errors seemingly associated with the  `/home/<user>/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c` file:

```
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-rmt.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-spi.c.o
/home/<user>/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/home/<user>/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/home/<user>/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-time.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-timer.c.o
```

Is this a problem with my PlatformIO environment, C install or a programming issue?  Can it be fixed?

Thanks

Roger

Bakafish, 2021-08-17:
You added the line I mentioned, or changed the existing config line already present to `dio`? The config line was already in that file, just with a different setting, you should not have it in there twice with conflicting values. 

I'm pretty sure the two 'Warnings' above are ignorable. They are coming from the platformio code, they will fix the warning at some point, but I think the underlying code still works as expected.

Make sure to do a `Clean` (click the trash can) and then update to the latest from the GIT since I think the Devs pushed out the latest Devt build to the release branch. 

Please be clear if you are still experiencing the crash loop at boot or not.

MitchBradley, 2021-08-17:
The warnings are from the Arduino framework code that is being compiled with more stringent warning settings in the Platformio compiler setup.  They can be ignored - we all get them.  To "fix" it you would have to complain to the people who maintain frameworkarduinoespressif32

Project487, 2021-08-17:
Thanks for the responses.

I deleted my grbl_esp32 git local repository and recloned.  I now have a successful test_drive machine and will hopefully will be able to create my own now.

closed 2021-08-17 by Project487