bdring/Grbl_Esp32 #957: Grbl_Esp32_Utility: A Graphical User Interface for Initial Setup

ajquick, 2021-08-16:
Over the weekend I wrote a GUI application for the initial configuration of Grbl_Esp32.

https://github.com/CNCxyz/Grbl_Esp32_Utility

[image]

The idea was that eventually end users may buy pre-installed devices without having to compile and install it themselves, so having a program that sets the initial Wi-Fi and access point settings easily would be a good idea. I didn't include all the settings, but may go back and add them. Obviously once on the network the main settings can be configured in the WebUI.

The program was built using Qt open source community software. In theory the program can be compiled for MacOS and Linux, but all I have is Windows to test with at the moment. If anyone wants to help out, improve the code, add features or help do some cross compiling, I would appreciate the help!

MitchBradley, 2021-08-18:
Another alternative would be to add a screen to WebUI as follows:, If WebUI is connected in AP mode or otherwise determines that the firmware does not have a proper STA configuration, it brings up a simplified screen to setup the STA configuration.  The idea is to use your phone to connect to the AP and do the initial setup.  The difficulty with the Qt solution is that it is yet another program that the user would have to install and the developers would have to maintain.  The program is relatively simple at some level, but dealing with a different code base on top of a complex framework, compiling on three different platforms, maintaining it, and supporting the inevitable user questions is never easy.  Every new component, especially one in a different environment, always comes with a lot of grunt work.

The advantages of using a phone for setup - assuming that there is a page that is laid out for easy of use on a small screen - is that nearly everyone has one handy at all times, and it is relatively easy to switch the wifi over to the ESP32 AP and back.  Switching a computer over can be more complicated due to the loss of internet connectivity; phones tend to be easier in that respect because they assume that wifi access to different networks will come and go - and you probably have cellular data as alternative internet access.

Another advantage of the phone method is that you do not need a USB cable or a driver for the USB serial chip.

ajquick, 2021-08-18:
The way I look at it, the project doesn't need to be updated beyond it's basic functions as long as no additional features need to be added. There are maybe a few more tweaks I want to make, perhaps adding Bluetooth support and DNS/Gateway settings, but that's it. You are correct that a web browser can be used, but if using from a phone the Index.html file still needs to be uploaded which can be hard to do from a phone. Of course that doesn't matter if it's already been done by the person that installed the firmware.

The GEU program is meant to be portable (no installation). It should be possible to get it down to a single EXE file that can be run from anywhere. Just a tidy project that can be stored on the SD card (if equipped) and run when needing to make changes (such as if your Wi-Fi password changes and you're stuck in STA mode).

> Another advantage of the phone method is that you do not need a USB cable or a driver for the USB serial chip.

While my ideal ESP32 setup would have multiple power options, barrel jack, screw terminals, POE and USB I think lots of setups would have their controller powered over USB and any OS newer than Windows XP should auto install the serial drivers.

MitchBradley, 2021-08-19:
I have never seen a project that did not need periodic updates.

The phone would not be used to upload index.html.gz, only to get onto the real network.  Uploading index.html.gz would be done via the normal WebUI methods.

A single .EXE file only works on Windows.  MacOS and Linux, of course, have different packaging issues.  Linux is especially annoying due to the many different distros and the PC vs Raspberry Pi axis.  If you are willing to own the project across the different platforms, go for it - but you have already indicated that you are not currently set up for anything other than Windows.

Grbl_Esp32 does not typically get stuck in STA mode; if it cannot connect in STA mode after a few tries, it falls back to AP mode.

Powering Grbl_ESP32 solely from USB is possible, but barely so.  Bart's controllers all depend on an external power source for most of the board.  You can power the ESP32 from USB, but once you add anything else, you are getting close to the limit of what a standard USB port can supply.  A board that talks to a few external drivers and little else can probably get away with just USB power, but that is not the bulk of the market that we have seen so far.  USB power is enough for initial setup though.

On the CNCjs User Group on Facebook, we frequently field questions from people who cannot figure out which serial port to use .  It's not a huge problem, but it does take up support time.

ajquick, 2021-09-02:
Thanks for your feedback Mitch. 

I believe the roadmap for the program as well as FluidNC is going to require more specialized GUI type setups in the near future. Helping fill that need is my aim for creating this *open source* project to support users of Grbl_Esp32. 

I'm not sure if I was demanding users assist in the development and periodic updates, I was merely stating that the code could be compiled for Linux and Mac. I have gone ahead and loaded Qt on Linux and Mac and have created a new Windows (exe), MacOS (dmg) and Linux GUI (appimage). Barring any bug fixes (or until FluidNC comes out) the programs should just *work*. I'm not sure if I gave off the impression that the project was immediately abandoned upon release or not, it was not my intention.

I have my own plans to use the firmware to create a controller that does in fact run off of USB. It does not have the stepper driver's built in, so the power consumption is not a concern. The end user will be able to use either USB direct or USB as power only and run over Wi-Fi. My intention here with the GUI is to create something that can be used directly for when the end user is not also the one compiling the firmware. This helps with the users that may not understand how to open the serial port and send commands as you have experienced with CNCjs. The simple program aims to be somewhat fool proof as it is able to check if Grbl is installed on the ESP32 before trying to change settings. Hopefully end users would switch to the next available port to try if the first one does not work.

Now that being said, running an EXE, DMG or Appimage (even a portable one) is an annoyance for the end user. That is why I have taken what I have written in Qt and indirectly ported it to a Web Serial program that exists entirely online. Now you can update your settings over serial using the latest versions of Chrome, Edge and Opera without needing to download anything.

This is a proof of concept. Additional work needs to be done to handle errors, provide more information and generally be more helpful to any end users that do have questions. The benefit here is that the files can be run locally, without needing NodeJs installed, as well as run over the web through Github's hosted pages. I will need to go through and clean up, my (poorly written) Javascript but it is currently working:

Source: https://github.com/CNCxyz/Grbl_Esp32_Web_Utility

Web: https://cncxyz.github.io/Grbl_Esp32_Web_Utility/