bdring/Grbl_Esp32 #959: Backlash Compensation

bderkk, 2021-08-24:
I try to find if there is a backlash compensation option but I could not find it. 
Is there any intention to add it?

I think that could be a really interesting option.

What do you think?

:)

MetalWorkerTools, 2021-08-29:
Depending on the direction of the load and the direction of the previous axis move, backlash compensation (BLC) should set or not. Because the controller does not know in what direction the loads comes, it can't compensate for backlash.

Imagine you are (**conventional**) milling in the Y direction and the workpiece is on the left side of the mill. The workpiece will push the mill to the right side. If the last X-axis move was to the left side, no compensation is needed.
The same situation but the workpiece is on the right side of the mill. The workpiece will push the mill to the left side and because the X-axis last move was to the left side, you have to move the X-axis to the right side to eliminate backlash.
Only you and the CAM processor know the direction of the load and can decide when to compensate.

shooter64738, 2021-09-01:
I ported backlash compensation from my control firmware over to grbl and put it on a mega2560 board. It should port fairly easily to this flavor of grbl as well. You set the amount of backlash to compensate using the $ parameters 140,141,142, etc... Compensation should happen during a direction change of the axis. 

[grbl with backlash comp](https://github.com/shooter64738/grbl-mega)