bdring/Grbl_Esp32 #961: only turn on wifi with usb cable connected how to solve

Alexandre2003, 2021-08-29:
it only turns on wifi with usb cable connected how can I solve it?

since already thank you very much

MitchBradley, 2021-08-29:
Does the board have power apart from the USB?

Alexandre2003, 2021-08-29:
yes I use external power I tried both for Vin with 5V and for 3v3 and the same happens the wifi only turns on if you connect the usb

MitchBradley, 2021-08-29:
We need to know everything about your system.  I have never seen a problem like this with Bart's boards, so we need to know about your board, the ESP module you are using, what is connected externally, which software version you are using and the contents of your machine file.

Alexandre2003, 2021-08-29:
I intend to do my own so I'm testing it on a protoborad , I'm using a 36-pin esp32 wroom DEVKIT-V1 module I'm isando to compile the IDE 1.8.13 on the protoboard I just connected an external driver tb6600 on the x-axis for testing which worked perfectly. also the functions of SPINDLE_ENABLE_PIN COOLANT_MIST_PIN and COOLANT_FLOOD_PIN through leds only for testing everything apparently worked perfectly only this wifi problem some time ago I had tested it with the much older version of the firmware and I didn't have this problem so I'm lost
____________________________________________________________
\ I downloaded it here I think it's the most up-to-date   /
  \                                                                                    /
    \         https://github.com/bdring/Grbl_Esp32         /
      \__________________________________________________/
follow below how the MACHINE file is

<machine.h>

#pragma once

// This file is where you choose the machine type, by including
// one or more machine definition files as described below.

#ifndef MACHINE_FILENAME

// !!! For initial testing, start with test_drive.h which disables
// all I/O pins
// #include "Machines/atari_1020.h"
//#    include "Machines/test_drive.h"

// !!! For actual use, change the line above to select a board
// from Machines/, for example:
 #include "Machines/3axis_v4.h"

// === OEM Single File Configuration Option
// OEMs that wish to publish source code that is configured for a
// specific machine may put all of their configuration definitions
// directly in this file, without including any other file above.

#else

// By using the external environment to define MACHINE_FILENAME,
// a configuration can be chosen without editing this file.
// That is useful for automated testing scripts.
//
// For example, when using the platformio compilation environment
// under Linux, you could issue the following command line:
//   PLATFORMIO_BUILD_FLAGS=-DMACHINE_FILENAME=3axis_v4.h platformio run
//
// Under Windows, using PowerShell, the command would be:
//   $env:PLATFORMIO_BUILD_FLAGS='-DMACHINE_FILENAME=3axis_v4.h'; platformio run
//
// When using the Arduino IDE, there is no easy way to pass variables
// to the compiler, so this feature is not useful for Arduino.
//
// MACHINE_FILENAME must not include the Machines/ path prefix; it is
// supplied automatically.

// MACHINE_PATHNAME_QUOTED constructs a path that is suitable for #include
#    define MACHINE_PATHNAME_QUOTED(name) <src/Machines/name>

#    include MACHINE_PATHNAME_QUOTED(MACHINE_FILENAME)

#endif  // MACHINE_FILENAME

Alexandre2003, 2021-08-29:
I even thought that the esp32 board is having problems so I bought another module I think it will arrive tomorrow

MitchBradley, 2021-08-29:
Is there an LED on the ESP module that comes on when there is power, and if so, is that LED on when there is no USB?

Alexandre2003, 2021-08-29:
yes the led is on when the usb is connected and it also lights up when i disconnect the usb and turn on the external power

MitchBradley, 2021-08-29:
Does the 3axis_v4.h machine config file reflect your hardware pin assignments?

Since we cannot duplicate your setup due to that custom hardware and the use of a 36-pin module, the only thing I can suggest at this point is to "bisect" by trying different firmware versions to discover the precise boundary between the last working version and the first non-working version.  The changes between those versions will give a strong clue about what could be failing.  To do such a bisection, you will need to use git to acquire the source code, instead of downloading a .zip.  With git, you can easily switch between versions.

Alexandre2003, 2021-08-29:
yes the 3axis_v4.h file works very well in relation to the pins I will try it with git

Alexandre2003, 2021-08-29:
@MitchBradley 

I was thinking that it could be some requirement for usb (serial) or something, so I decided instead of connecting the usb cable to the computer, connect to a usb source (cell charger) and the wifi turned on and connected to my wifi network normally and accepts all the grbl command but if I remove the usb the wi fi turns off which makes me think that the board is having problems I will wait for the new one to arrive and test

MitchBradley, 2021-08-29:
There are two possibilities that I can think of.  Either the ESP chip is not getting power, despite the fact that the LED is on, or the serial chip is holding the ESP chip in reset - or going into programming mode - when there is no power directly on the USB port.  The serial chip controls the EN line for the purpose of resetting the chip into programming mode.  The 38-pin DEVKIT modules use a CP2102 serial chip that works correctly for reset and programming.  I have seen modules that are advertised to use different serial chips.  Who knows how they are wired and how they behave when there is no USB power?

Alexandre2003, 2021-08-29:
So the module I'm using has a CP2102 serial chip, tomorrow the new module should arrive, I'll test it and post the result here

MitchBradley, 2021-08-29:
You can look at the pins that control booting with a multimeter.  
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

Alexandre2003, 2021-08-31:
@MitchBradley 

Good night friend solved the problem!! look what happens for esp32 to initialize correctly by the V-IN input I had to increase the voltage on the V-IN instead of sending 5V I had to increase it to 5.7V and the board correctly initialized the wifi and the motors and peripherals controls are all OK thank you very much for your help a big hug

MitchBradley, 2021-08-31:
Maybe there is a protection diode between the 5V and the linear regulator, and perhaps it is an ordinary diode with a 0.7V drop instead of a Schottky diode with a 0.2 to 0.3V drop.

closed 2021-08-31 by MitchBradley

Alexandre2003, 2021-08-31:
this is very likely