bdring/Grbl_Esp32 #962: No reading of the SSD card in the WebUI

RomRas, 2021-08-30:
I have built a MidTBot ESP32 plotter described at: https://github.com/bdring/midTbot_esp32.
Everything is OK except reading the micro SD card in the WebUI panel.
Why is the content of the SD card not displayed?
The serial interface reads the content of the SD card correctly after sending the command "$ SD / List"!
1 - Controller midTbot ESP32 V2
2 - Boot Messages:

[MSG:Grbl_ESP32 Ver 1.3a Date 20210816]
[MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:Using machine:midTbot]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:Global stepper disable pin:GPIO(13)]
[MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(26) Disable:None Limits(5.000,105.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(25) Disable:None Limits(-100.000,0.000)]
[MSG:Z  Axis RC Servo Pin:27 Pulse Len(6553,3276) Limits(0.000,5.000)]
[MSG:Servo Update Task Started]
[MSG:CoreXY (midTbot) Kinematics Init]
[MSG:No spindle]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:X  Axis limit switch on pin GPIO(2)]
[MSG:Y  Axis limit switch on pin GPIO(4)]

Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]

3 - Settings:

$Z/RcServo/Cal/Max=1.000
$Z/RcServo/Cal/Min=1.000
$Sta/SSID=GRBL_ESP
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=GRBL_ESP
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=grblesp
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=btgrblesp
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=Y
$Homing/Cycle2=X
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=3
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=XYZ
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=Off
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=XZ
$Homing/Squared=
$Homing/Feed=500.000
$Homing/Seek=2000.000
$Homing/Debounce=250.000
$Homing/Pulloff=3.000
$GCode/MaxS=1000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=NONE
$X/StepsPerMm=100.000
$Y/StepsPerMm=100.000
$Z/StepsPerMm=100.000
$A/StepsPerMm=100.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=8000.000
$Y/MaxRate=8000.000
$Z/MaxRate=5000.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=200.000
$Y/Acceleration=200.000
$Z/Acceleration=100.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=5.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=100.000
$Y/MaxTravel=100.000
$Z/MaxTravel=5.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=0.250
$Y/Current/Run=0.250
$Z/Current/Run=0.250
$A/Current/Run=0.250
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.125
$Y/Current/Hold=0.125
$Z/Current/Hold=0.125
$A/Current/Hold=0.125
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=16
$Y/Microsteps=16
$Z/Microsteps=16
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
ok

4 - Nie dokonałem żadnych zmian w GrblESP32

5 - Połączenie przez WebUI

Alexandre2003, 2021-09-04:
@RomRas 

Good morning RomRas I had the same problem that was solved by making a change in line 122 of src / Serial.cpp of

                            4096, // size of task stack
for

                             8192, // size of task stack

RomRas, 2021-09-04:
Hello,
Thank you very much for the information sent.
I have installed the ESP32 library version 1.0.3 in the board manager in the Arduino IDE. Originally, I had version 1.0.6 installed. Everything is OK now!

Roman
---------------------------------------------------------------------------

From: Alexandre2003 
Sent: Saturday, September 04, 2021 11:25 AM
To: bdring/Grbl_Esp32 
Cc: RomRas ; Mention 
Subject: [SPAM] Re: [bdring/Grbl_Esp32] No reading of the SD card in the WebUI (#962)

@RomRas

Good morning RomRas I had the same problem that was solved by making a change in line 122 of src / Serial.cpp of

                        4096, // size of task stack
for

                         8192, // size of task stack
—

Ta wiadomość została sprawdzona na obecność wirusów przez oprogramowanie antywirusowe Avast.
https://www.avast.com/antivirus