bdring/Grbl_Esp32 #963: Easiest ESP32 board build for small CNC router (with rotary axis)?

egfreelance, 2021-09-03:
Hey guys. I've been trying to build a CNC and I've been having a really hard time with all this board stuff. I need a rotary axis compatible setup, which kills most of my easy options. I could just get the xPro V5 but it sounds like I could make the same thing or better for less $$$. But that depends on the level of DIY I have to do. I'm not very good with all this wiring and Arduino stuff. Is it possible to get something like [this](https://www.amazon.com/BIGTREETECH-Upgrade-Control-TMC2209-Creality/dp/B0882QGFZR/) and flash it with something I can download and plug in steppers and power and have a working setup, or is there a lot more effort into getting it working right? I've also been looking into the MakerFR board, any idea how hard that is to get working? I'd rather not have to shell out and get the xpro but if I need to do lots of custom code or anything that sounds like it may be too hard and I should just cough up the money haha. Thanks for any help.

MitchBradley, 2021-09-03:
Based on all that you have said about having a hard time, you would probably be better off paying the extra money to get something with a lot of support from the vendor.  You asked the question "is it possible" - to which the only correct answer is "yes, if you or someone else has the necessary skill and is willing to spend the time".  Since you have said that you do not have the skill, a better question would be "how can I entice someone with skill to help me".  If you choose to make your own - which seems challenging given your self-described lack of skill - there is nobody with any particular incentive to help you.  If  you purchase something, the vendor might feel obliged to help.  There is often a strong correlation between the $$$ you spend and the vendor's willingness to help.  If you buy the cheap Chinese board, maybe - just maybe - you can find a forum somewhere where some random person might help you, but you won't get any help from the vendor.

MitchBradley, 2021-09-03:
The xPro is a nice package, but if I were you, I would try to contact the vendor to gauge their willingness to support you.  That goes for any route you choose; if you will need help, you should try to determine in advance how likely you are to get it.

egfreelance, 2021-09-03:
Hey Mitch, thanks for the reply. I fully understand what you mean on that there's no reason for someone to go out of their way to develop all this for me. Which is true. My thought was more that a lot of these open source projects usually have things that someone has already done and set up that they decide to share. The first thing that comes to mind is 3d printers. When I upgraded my extruder on my old solidoodle 4 I had to flash the board with new stuff, but I didnt have to make any of it or anything special. It was supplied on the upgrade guide on the forum, which I was able to follow and ended up with a nice upgraded printer. So I was wondering if there's anything like that out there that I haven't found. 

On support, I feel like once I have somethign set up I shouldn't need much help. I generally just export g code from fusion 360 and run it in my grbl program (I use candle but plan to check out the others). But it is a thing I should keep in mind just in case I run into issues. I'll email spark concepts and ask how hard it is to set up for rotary axis.

MitchBradley, 2021-09-03:
If you want help from a collection of users, one of whom might have something similar to yours, it would be better to ask on a forum.  Issues on a github project tend to be read mostly by developers.

closed 2021-09-03 by MitchBradley