bdring/Grbl_Esp32 #964: No PWM output on MPCNC V1.2.2 (PWM Spindle)

tchleb, 2021-09-09:
Hello,

i compiled (5316c64) and flashed the GBRL_ESP32 on a MPCNC V1.2.2 Board. I have an issue that i don't have PWM signal on GPIO16 (3v3 Level) when i try to switch on the spindle over the webinterface. I checked this with an scope.

The spindle enable (GPIO32) is not set to high for me either

I use the same [DevBoard](https://camo.githubusercontent.com/8ffa91d918dcb0e0e04dec40c6f09554d715055c2d6bf9806dd263851f3dd973/687474703a2f2f7777772e6275696c646c6f672e6e65742f626c6f672f77702d636f6e74656e742f75706c6f6164732f323031382f30382f4553502d33322d4465762d426f6172642d383030783830302e6a7067) which are shown on the wiki.

Does anyone have a hint or had such an error already?

br 
Tobias

_____________________________________________________________________________________________________________________________________
mpcnc_v1p2.h
_____________________________________________________________________________________________________________________________________
#pragma once
// clang-format off

/*
    mpcnc_v1p2.h
    Part of Grbl_ESP32

    Pin assignments for the Buildlog.net MPCNC controller
    https://github.com/bdring/Grbl_ESP32_MPCNC_Controller

    2019    - Bart Dring
    2020    - Mitch Bradley

    Grbl_ESP32 is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    Grbl is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with Grbl_ESP32.  If not, see <http://www.gnu.org/licenses/>.

    ## Stepper Driver Pins
    # |-------|------------|----------|---------|--------------|
    # | MOTOR | ENABLE pin | STEP pin | DIR pin | ENDSTOP pin  | 
    # |-------|------------|----------|---------|--------------|
    # |  X    |	  GPIO13   |  GPIO12  |  GPIO26 |    GPIO17	   | 
    # |-------|------------|----------|---------|--------------|
    # |  X2   |	  GPIO13   |  GPIO22  |  GPIO26 |    GPIO17	   |
    # |-------|------------|----------|---------|--------------|
    # |  Y    |	  GPIO13   |  GPIO14  |  GPIO25 |    GPIO04	   |
    # |-------|------------|----------|---------|--------------|
    # |  Y2   |	  GPIO13   |  GPIO21  |  GPIO25 |    GPIO04	   |
    # |-------|------------|----------|---------|--------------|
    # |  Z    |	  GPIO13   |  GPIO27  |  GPIO33 |    GPIO15	   |
    # |-------|------------|----------|---------|--------------|

    ## Other inputs and outputs
    # |---------------|---------|
    # |    Function   |   PIN   |
    # |---------------|---------|
    # |    Hold       |  GPIO36 |
    # |---------------|---------|
    # |    Start      |  GPIO39 |
    # |---------------|---------|
    # |   GBRL_Reset  |  GPIO34 |
    # |---------------|---------|
    # |    Probe      |  GPIO35 |
    # |---------------|---------|
    # |  Spindle_EN   |  GPIO32 |
[... 197 lines cut ...]
[MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(26) Disable:None Limits(0.000,500.000)]
[MSG:X2 Axis Standard Stepper Step:GPIO(22) Dir:GPIO(26) Disable:None Limits(0.000,500.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(25) Disable:None Limits(0.000,500.000)]
[MSG:Y2 Axis Standard Stepper Step:GPIO(21) Dir:GPIO(25) Disable:None Limits(0.000,500.000)]
[MSG:Z  Axis Standard Stepper Step:GPIO(27) Dir:GPIO(33) Disable:None Limits(-80.000,0.000)]
[MSG:PWM spindle Output:GPIO(16), Enbl:GPIO(32), Dir:None, Freq:5000Hz, Res:13bits]

[MSG:Client Started]
[MSG:Connecting myWifi]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connected with 192.168.178.87]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(2)]
[MSG:X  Axis limit switch on pin GPIO(17)]
[MSG:Y  Axis limit switch on pin GPIO(4)]
[MSG:Z  Axis limit switch on pin GPIO(15)]
[MSG:Probe on pin GPIO(35)]

Grbl 1.3a ['$' for help]

bdring, 2021-09-09:
Try sending M3 S1000 manually.

If that works, try using the WebUI buttons. The spindle needs a speed value before M3 does anything. Sending Sxxxx where xxx is the RPM anytime will set the Rpm .

You can check the current rpm with the $G command.

tchleb, 2021-09-10:
Thank you for your reply.

I found my issue...

The level converter must have broken due to an electrical surge on the 5v rail.  This pulled gpio16 & 32 to ground.  After I swapped the level converter, the pwm output works

closed 2021-09-10 by bdring