bdring/Grbl_Esp32 #973: Problems Compiling Firmware

281Er, 2021-10-01:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?

What version of the Arduino IDE are you using? 

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 

Are you using the master branch of Grbl_ESP32?

Have you made any edits or configuration changes (list them) to the  firmware?

Please paste the compiler error text here:

I am running the cnc xpro v5 and I am having issues with long programs. Usually when I run a program longer that 30 minutes it ends up stopping mid program. I'm running over webui on grbl, and it just goes to idle on the top right where you are able to pause program if needed. From my understanding once I send the program to the controller it goes to the sd card, so it shouldn't be an issue of the laptop. The program is perfect no issues there, I'm using aspire. Could it be an issue with Grbl? If anyone has any insight I would appreciate it.

MitchBradley, 2021-10-01:
The title of this issue has nothing at all to do with the problem description.

281Er, 2021-10-01:
It seems formatting the sd card might have done it. If the problem persists I'll upgrade the sd card.

closed 2021-10-01 by MitchBradley