bdring/Grbl_Esp32 #974: I/O voltage levels

mechapro, 2021-10-03:
Hi,

I'm wondering how the I/O voltage levels of ESP32 and motor drivers and other external components fit together? As far as I see, ESP32 works in the 3V3 domain, while the I/O extension boards offer Vmot and 5V only? I didn't find any warnings to use level shifters?

Best Regards
Thorsten

bdring, 2021-10-03:
The ESP32 is a 3.3v part. None of the controllers input more the 3.3V to the ESP32. Most stepper driver modules are OK with 3.3V as the input voltage for step direction, etc. Some external drivers with opto inputs require more than 3.3V. Level shifters should be used with those. modular controllers like the 6 Pack have a lot of options for outputs.

mechapro, 2021-10-03:
OK, got it. Thanks.

closed 2021-10-03 by bdring

MitchBradley, 2021-10-03:
1) Most optocouplers actually work fine with 3.3V
2) The small bidirectional level shifters that are used for converting CMOS logic signals will not work for this because they are not designed for high current.  Optocouplers need on the order of 10 mA to turn on - although they will often work down to 5mA or perhaps a bit less.