bdring/Grbl_Esp32 #979: Another homing issue.

Wankah, 2021-10-13:
I know theres tons of topics on homing issues. Ive read most of them ive browsed wikki up and down so many times. Please dont let me go back to the mega 2560 version. 

Firmware set for NO switches, pins arent inverted (inverted them just for fun i guess) no notable EMI or false trigger issues. Using a filter on the limits. Ganged the Y axis was trying to use self squaring $HY sounds no alarms. bumps both Y motors at once then bumps Y1 2 times bumps Y 2 times, replies with ok. Do i need to add something into my machines.h folder? Any tips on a .h to look into? Ive looked through every file several times to see if ive missed something. I bumped the terminal into 1 second updates and watched for over an hour and never saw a single limit trigger since i put in my filter.

bdring, 2021-10-13:
Are you saying it homes fine, but touches the switches more times than expected?

It should move until one switch touches. It then needs to touch both switches one more time each. It does this because both Y and Y2 switches are treated the same internally. It does not know which one is getting touched. It just infers it from the motor that is moving.

Having 2 inputs was complicated in a port of Grbl. We wanted to wait on that for the next generation. We now have [FluidNC](https://github.com/bdring/FluidNC) in beta testing where the switches are separate and it can do a simpler squaring sequence.

Wankah, 2021-10-13:
No they only move what i have the bounce back set at. Forget what that grbl setting is called. I know what its doing. Ish. Its acting like the limit is already triggered and its trying to come off the limit. My only problem with fluid is honestly i have no idea what im doing with this or fluid you guys do an awsome job with the wiki .it gets you in the ballpark.