bdring/Grbl_Esp32 #980: MQTT support

willertan, 2021-10-13:
This is the perfect project for linking smart home (or smart factory) and cnc machine.
I wish it has MQTT support to publish status to MQTT server so I can monitor everything from my smart home dash board or trigger something like vacuum/air ventilator during CNC milling.

Currently I am using these for my Home Assistant (https://www.home-assistant.io/)
3D printer to smart house integration - https://www.home-assistant.io/integrations/octoprint/
CNCJS to home assistant integration - https://www.npmjs.com/package/cncjs-mqtt

MitchBradley, 2021-10-14:
How much other stuff is necessary before this can be used?  How standardized is it?  I know that the MQTT protocol itself is quite standard, but what about the brokers and clients?  How much other stuff is necessary in order to make use of MQTT signaling and how many different versions of that other stuff is out there?  What sort of support burden would we be taking on when people who are new to the home automation space start wrestling with different HA platforms?  I know of at least 16 open source HA platforms and perhaps 9 proprietary ones.

MitchBradley, 2021-10-14:
Brokers are interchangeable as far as function is concerned, but not from the standpoint of installation and administration.  It is those aspects that generate an outsized amount of support hassle.  Everybody wants to use a different setup.