bdring/Grbl_Esp32 #988: how can i set setting manualy in code?

Rezaasghariir, 2021-10-31:
hi
I have this setting 
    table_feed_rate         = new IntSetting(EXTENDED, WG, NULL, "SandTable/TableFeedRate", DEFAULT_TABLE_FEED_RATE, DEFAULT_TABLE_MIN_FEED_RATE, DEFAULT_TABLE_MAX_FEED_RATE,tablefeedrateChange);
how can i change this setting in code ?

bdring, 2021-10-31:
I don't know what you mean. Why would you have a setting and change in code?

Explain exactly how this is used? What does the setting do? Feed rates usually come from gcode.

Rezaasghariir, 2021-10-31:
my feedrate is constant and not change from gcode but i want change that with slider from WebUI and Web server handlers.
in previous version I used code like this settings_store_global_setting(37, value); but in last version how can i do that?

bdring, 2021-10-31:
I suggest: Put an average feedrate in the gcode and use the existing feedrate override feature. It is already in the WebUI

You can change any setting by sending the dollar command 

**$SandTable/TableFeedRate=XXX.XX**

Rezaasghariir, 2021-10-31:
i have my WebUI And new design.
i need change feedrate when SDCARD is in printing and can't use $ command.
is any way to set setting rather than $ command?

MitchBradley, 2021-10-31:
The only way to change something during a file run is to send a realtime command character.

Changing a setting value during a run is a very very bad idea.  Settings are stored in FLASH so that they persist across reboots.  Writing FLASH is very slow because of the way the ESP32 is designed, and it causes other code execution to pause.

The way to do what you want is to have the setting store the initial rate, then read that rate into a variable that can be adjusted with a realtime command character, in the same way that feedrate and spindle speeds are adjusted.

Rezaasghariir, 2021-11-01:
tanks for your answering.
i was edit this function ConnectSTA2AP. thats improve connect to station how can i send that for you ?
here is my code:

    bool WiFiConfig::ConnectSTA2AP() {
        String      msg, msg_out;
        uint8_t     count  = 0;
        uint8_t     dot    = 0;
        uint8_t reconnect = 0;
        wl_status_t status = WiFi.status();

        while (status != WL_CONNECTED && count < 40) {
            if(reconnect>7){
   
                    String password = wifi_sta_password->get();
                    String SSID     = wifi_sta_ssid->get();
                    WiFi.begin(SSID.c_str(), (password.length() > 0) ? password.c_str() : NULL);
                    reconnect = 0;
            }
            switch (status) {
                case WL_NO_SSID_AVAIL:
                    msg = "No SSID";
                    break;
                case WL_CONNECT_FAILED:
                    msg = "Connection failed";
                    break;
                // case WL_DISCONNECTED: {
                //     String password = wifi_sta_password->get();
                //     String SSID     = wifi_sta_ssid->get();
                //     WiFi.begin(SSID.c_str(), (password.length() > 0) ? password.c_str() : NULL);
                // } break;
                case WL_CONNECTED:
                    break;
                default:
                    if ((dot > 3) || (dot == 0)) {
                        dot     = 0;
                        msg_out = "Connecting";
                    }
                    msg_out += ".";
                    msg = msg_out;
                    dot++;
                    break;
            }
            grbl_sendf(CLIENT_ALL, "[MSG:%s]\r\n", msg.c_str());
            COMMANDS::wait(500);
            count++;
            #ifdef SAND_TABLE
            reconnect++;
            #endif
            status = WiFi.status();
        }
        return status == WL_CONNECTED;
    }

MitchBradley, 2021-11-01:
The proper way to submit code is to create a github Pull Request.  There is a lot of documentation about pull requests on the Internet.  We have some additional guidelines at https://github.com/bdring/Grbl_Esp32/wiki/Pull-Request-Guidelines .

Please be advised that changes to Grbl_Esp32 are likely to be ignored because we have switched over to working on FluidNC, which is based on Grbl_Esp32 but already superior to it in nearly every way.