bdring/Grbl_Esp32 #989: Just a question regarding pinout and how to assign in code

mac7988, 2021-11-01:
I made a custom board which has a lot of Bart's schematics implemented.  I just don't understand how to implement a custom machine.  Below is what I have and any help would be really good. I see some examples of custom machines. I just don't know how I can make my own file and how to tell the ESP which pin would do what. Some examples would help.

Thank you in advance.
To simplify lets discuss only X axis.

X limit sensor is on IO33
[image]

X Stepper is using a 74HC595D 
[image]

2 relay output are from IO15 and IO14
[image]

I have 3 inputs from IO4, IO16, IO26
[image]

bdring, 2021-11-01:
If you are just getting started, I would suggest learning FluidNC. It is the next generation of Grbl_ESP32.

bdring, 2021-11-01:
See this [Wiki](https://github.com/bdring/FluidNC/wiki)

Here is an invite to our [Discord server](https://discord.gg/JWftWRsV).

mac7988, 2021-11-01:
Thank you guys, is this stable?

MitchBradley, 2021-11-01:
I am using it on my production machine.

MitchBradley, 2021-11-01:
As far as we know it is strictly better than Grbl_Esp32, with the sole exception that there is some custom code for oddball machines that has not yet been ported over.  By oddball, I mean strange kinematics and support for unusual motors.  Everything related to cartesian machines using stepper motors with various drivers is at par with, and in some cases ahead of, Grbl_Esp32.

MitchBradley, 2021-11-01:
The other point is that we are actively working on FluidNC so any bugs that are found - especially critical ones - get fixed very quickly.  Grbl_Esp32 maintenance is not happening as a result of our narrow focus on FluidNC.

mac7988, 2021-11-01:
I understand. If you were to "tune" for my configuration then how would it look? an example would help.

MitchBradley, 2021-11-01:
We already told you.  Look at the pointer in my first response.

bdring, 2021-11-01:
Most of my controllers have example configs.

https://github.com/bdring/FluidNC/tree/main/example_configs

closed 2021-11-01 by MitchBradley