bdring/Grbl_Esp32 #992: Status Report on Telnet is not working

piyushverma, 2021-11-09:
Hello,

I am able to connect using telnet and send command, I also get reply ok\r\n.

But I am not able to get report status even when $10=2 is set, How ever status update is reported on webui.

Any suggestion ?

Thanks
Piyush

MitchBradley, 2021-11-09:
You usually have to send an Enter after ? on most Telnet clients, because telnet is line oriented by default.  Some telnet clients can be put into a character-at-a-time mode, but they often do not work very well in that mode.  The Microsoft telnet client has "stream" and "console" modes.  I forget which is which, but one is line oriented and the other is character oriented.  Consult the documentation for whichever Telnet client you are using to learn more.  Note that the Grbl_Esp32 telnet feature is not true Telnet, but rather a raw TCP stream that happens to be, by default, on the Telnet port.  True Telnet has a negotiation phase in which the client and the server agree on various modes having to do with line endings and other things.  Grbl_Esp32's "telnet" does not implement that negotiation phase, it just shuttles characters back and forth over TCP.

closed 2021-11-09 by MitchBradley

piyushverma, 2021-11-10:
Thanks Bradley,

I was expecting automatic update but the fact is I need to send ? ,
I am making new CNC Controller App, I am connecting to telnet using socket.

Thanks again your information helped me to move forward.

MitchBradley, 2021-11-10:
Please be advised that work on Grbl_Esp32 is ceasing.  The developers are focusing all their efforts on the next generation FluidNC.

piyushverma, 2021-11-10:
Thanks I have noticed that, In free time I will try to build and use FluidNC.