bdring/Grbl_Esp32 #998: Realtime Command should be handeled wihtout \r\n

piyushverma, 2021-11-18:
Hello,

I am facing similar issue https://github.com/gnea/grbl/issues/327
I try to send realtime command ? wihtout \r\n but grbl does not respond anything.

If I add \r\n then it reply with 2 extra "ok" "ok"

even sending any grbl command like G0 G90 X0 cause double reply of 'ok' 'ok'

does It shoulds like a bug ?

MitchBradley, 2021-11-18:
The problem is not with grbl_esp32 but instead with the telnet client program on the host system.  Telnet clients usually start in "linemode", in which a line of data is collected and sent when the user types Enter (carriage return).  The telnet client program does not know that ? is a special Grbl realtime character so it treats ? just like any other printable character and collects it into a line, sending that line when the user types the end-of-line key.  The Telnet protocol was developed long before Grbl was written, and Grbl's use of ? as a special "send it now" character is quite unusual in the computer world.

Some telnet clients can be configured to use character-at-a-time mode.  Every telnet client has a different way of doing that, so you will have to consult the documentation for your telnet client program to learn how.

closed 2021-11-18 by MitchBradley

karoria, 2021-11-18:
Hi Piyush,
I think it replys \r and \n individually. So command and \r is treated one line, and \n is treated another blank line. I suggest to try only \n. It should return only one "ok". However, I doubt the realtime commands will work this way. Try websockets instead of telnet. If your http port is 80, then websocket port is default to 81. However, I am not successful communicating with ws for grbl_esp32. So I would like to know the cause if you succeed.
Best,
Ravi

karoria, 2021-11-18:
Oh sorry. It should not give any "ok" response. Should just reply with report in chevrons.
If you are on discord, join this project's discord server. Lots of discussions happening there to keep us with latest developments. Invite link is in wiki of fluidNC.

piyushverma, 2021-11-19:
Hi Ravi,

I can't find link to discord, it just say "There is a Discord server for the development this project. Ask for an invite"
I search for fluidNC but no server found.

Please suggest right name.

Thanks
Piyush

piyushverma, 2021-11-19:
> The problem is not with grbl_esp32 but instead with the telnet client program on the host system. Telnet clients usually start in "linemode", in which a line of data is collected and sent when the [...]

In that case \r\n should be consider transport layer data which should be deleted on telnet layer before passing to GRBL.

How ever I am using TCP Socket so it not depend on waiting for enter, but possible TELNET client side waiting for line termination in GRBL.

karoria, 2021-11-19:
> Hi Ravi,
> 
> I can't find link to discord, it just say "There is a Discord server for the development this project. Ask for an invite"
> I search for fluidNC but no server found.
> 
> Please suggest right name.
> 
> Thanks
> Piyush

Here is the link
https://discord.gg/MDsRDeNsTE

karoria, 2021-11-19:
It is in the wiki section where table of contents is mentioned. If u browse through PC, it is right side pane. In mobile it is at the end of page.

MitchBradley, 2021-11-19:
> In that case \r\n should be consider transport layer data which should be deleted on telnet layer before passing to GRBL.

That is not correct.  The line termination is part of the Grbl protocol.  If your TCP socket is not sending the '?' until it sees a line ending you need to fix its buffer flushing policy.

In any case, we are not likely to make any mods to the telnet code in Grbl_Esp32.  All of our development efforts have switched over to FluidNC.  I know for a fact that FluidNC handles '?" immediately if the Telnet client is in character mode.  I tested it yesterday.

piyushverma, 2021-11-20:
Thanks Ravi, I am in,

Thanks Mitch, I will build and try FluidNC

MitchBradley, 2021-11-20:
You do not have to build FluidNC, just download and install a prebuilt release.  All the configuration is done at runtime with a text file.