bdring/Grbl_Esp32 #999: Hardware Boards

pp-mpcnc, 2021-11-18:
Hey there,
A tech noob here...
I'm working on a 4th axis on my mpcnc wich is now running an arduino mega with ramps1.4. Since I'd love to keep the homing function I'd need another stepper driver. All the boards available with 6+ drivers are running 32bit chips. So my question is, how versatile is the esp32 code? What would be the problems I'd run into if I try to compile the code to an Bigtreetech GTR 1.0 or Octopus? Is it even possible?
Thanks for the help!

MitchBradley, 2021-11-18:
The esp32 code only works on esp32.  It has not been ported to ARM, and doing so would require a substantial amount of work.

You might consider this https://www.tindie.com/products/33366583/6-pack-universal-cnc-controller/ .  It is the board that is most heavily supported for grbl_esp32.

pp-mpcnc, 2021-11-18:
Thank you for the answer!

closed 2021-11-18 by MitchBradley