bdring/Grbl_Esp32 #1002: setting_codes_en_US.csv incomplete

markthepiggy, 2021-11-21:
Hi,

Grbl_Esp32 reports settings that do not appear in "setting_codes_en_US.csv".

These inlcude:

$103=100.000
$104=100.000
$105=100.000
$113=1000.000
$114=1000.000
$115=1000.000
$123=200.000
$124=200.000
$125=200.000
$133=300.000
$134=300.000
$135=300.000

MitchBradley, 2021-11-21:
That file is no longer being maintained, and the numbered settings are essentially obsolete in light of the new settings framework that is documented at https://github.com/bdring/Grbl_Esp32/wiki/Settings .  That new framework is itself being superseded by the full runtime configuration mechanism of FluidNC.  Grbl_Esp32 maintenance is winding down as the developers concentrate on FluidNC.

closed 2021-11-21 by MitchBradley

markthepiggy, 2021-11-22:
OK. I'll take a look at FluidNC. In the mean time, do you know what these settings are for?

MitchBradley, 2021-11-22:
Uh, read the link that I posted?