bdring/Grbl_Esp32 #1003: Problems Compiling Firmware

george20, 2021-12-08:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)? Yes

What version of the Arduino IDE are you using?  1.8.13

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 
v4.4-dev-3569-g6a7d83af19-dirty

Are you using the master branch of Grbl_ESP32? Yes

Have you made any edits or configuration changes (list them) to the  firmware? No

Please paste the compiler error text here:

#include <FreeRTOS.h>ResolveLibrary(FreeRTOS.h)

          ^~~~~~~~~~~~
  -> candidates: []compilation terminated.

My question is where can I find the FreeRTOS.h  and task.h  etc  called for in the Grble_ESP32 code?
as I understand it ESP32 uses  freertos to switch between its tasks.

MitchBradley, 2021-12-08:
Try just removing the "#include <FreeRTOS.h>" line from I2SOut.cpp .  It is probably redundant with a similar include that is already part of Arduino.h

We are not really supporting Arduino IDE compilation any more, having switched to PlatformIO.  Grbl_Esp32 support is also waning in favor of FluidNC.

benjamint76, 2021-12-10:
I encountered the same error in the Arduino IDE and opted for the PlatformIO route. This route was flawless and pretty quick!

george20, 2021-12-10:
Thanks for that. I encountered more errors so gave up! So i moved to FluidNC. Which compiled ok.

closed 2021-12-10 by george20

MitchBradley, 2021-12-10:
Why did you need to compile FluidNC? It is intended to be usable by simply downloading the precompiled installer package.

george20, 2021-12-10:
The short answer is i didnt know any different.
First off i havnt used platformio before.
So i downloaded the zip file, then built and uploaded it.  And it worked.
The wifi syntax took abit of trial and error, but eventually it connected
to my network.

Where would i find the pre built binary?

I havnt worked out how to upload the file system etc.
In parallel with this i am building a plasma cutter from scratch.
I bypassed arduino & R aspi controllers and went for ESP32. The latter I
have used for some years now.
So i am a complete newbie to the CNC world , and to your code in particular.
PS i am in my 7th decade.
George

MitchBradley, 2021-12-10:
If  you use the precompiled binary you don't have to deal with platformio at all.  Only us compiler jockeys have to worry about that stuff. [ Here is where you can find release binaries. ](https://github.com/bdring/FluidNC/releases) .  You can get there from the main page of the FluidNC by looking over at the right side and clicking on the "Releases" link.

The installer script that you get with the release bundle knows how to upload the initial file system.  Read the HOWTO in the release file.

george20, 2021-12-10:
Thanks
I will give it a go