bdring/Grbl_Esp32 #1007: GRBL 1.3a

DawidSacha1995, 2021-12-20:
Hello!
I have recently had an XPRO V5 CNC controller in my laser plotter. It has installed GRBL 1.3a( GRBL_ESP32) specially under ESP 32 microprocessor .
Unfortunately it doesn't work properly, Lightburn doesn't detect it, and manual configuration as grbl causes errors and stop of engraving process.
I tested other programs and with them this controller also does not work. I suspect errors in software 1.3a.
Please help me what code should I upload to make XPRO V5 work properly and preferably with LightBur

MitchBradley, 2021-12-20:
Have you contacted the xpro v5 vendor?  Presumably they took your money; if so they should provide some support.

DawidSacha1995, 2021-12-20:
Yes . They are not responding to my emails and the form on the Spark Concepst website which is very strange to me
I put in a lot of money and had orders for ornaments that are gone :(

Spark-Concepts, 2021-12-21:
Hi Dawid - sorry we havent been linked on email (ours are getting auto-spammed a lot lately). I dont want to clog up the ESP32 main page with our hardware specific issue, so please send an email to me directly - mike@spark-concepts .com or open an issue on our github page (https://github.com/Spark-Concepts/xPro-V5/issues) and we will get your sorted.  (likely you need to turn on hardware flow control in lightburn so the controller reboots on connection though).

DawidSacha1995, 2021-12-21:
Thank you for your reply 
I just removed my controller from the machine and am trying to change to new firmware (https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware).
LightBurn does not see my driver and I have to create it manually. When I do some g-code the machine makes strange movements outside the area where it should engrave until it stops with error 25. Also sometimes during jog mode it can hang up and you have to re-communicate. 
I tried the G-code in LightBurn but there, despite completing the entire program code laser is skipped areas that I want to engrave .
In CNCjs communication and settings are fine. In the attachment I send the settings.
I am helpless !  

[Settings.txt](https://github.com/bdring/Grbl_Esp32/files/7758183/Settings.txt)
[image]

MitchBradley, 2021-12-22:
Error 25 is "GCode Word Repeated".  That means that a GCode line was seen with a duplicate word, like for example G0 X1 X3 - but what it usually indicates in practice is a serial communications error that caused loss of characters.  The end of one line is lost and two adjacent lines are spliced together forming one invalid line.  We used to see that a lot because of a bug in the Arduino library serial driver.  That bug has since been fixed, and we also wrote our own Uart driver to avoid similar problems in the future.  Whether or not that bug is affecting the XPro firmware depends on when it was forked from our tree and which version of the Arduino framework was used to compile it.

MitchBradley, 2021-12-22:
FYI, I took the machine.h config file from the XPro tree and ran it through our automatic conversion program to generate a FluidNC YAML config file.  The result is at #1009